bitaxeorg/ESP-Miner #444: Add the expected Hashrateinfo  to the Frequency Tab on Settings

Bitcoinite, 2024-11-02:
Since we know the expected Hashrate for each Frequency (it is even shown on the Dashboard), we should also be able to add this info to Frequency dropdown on the Settings page. This way, users could see what the expected Hashrate for each Frequency setting is, without having to test every setting manually.

[image]

skot, 2024-11-02:
I think technically we could do this. I do worry that it will cause new users to attempt overclocking without understanding the risks (prompting more "my bitaxe is broken" support requests)

WantClue, 2024-11-02:
> I think technically we could do this. I do worry that it will cause new users to attempt overclocking without understanding the risks (prompting more "my bitaxe is broken" support requests)

I agree on that, technically this might be a nice thing, but increasing the interest in overclocking is probably not what we want to achieve.

closed 2025-04-16 by WantClue