bitaxeorg/ESP-Miner #447: First time setup - Change the message on OLED

STSMiner1, 2024-11-03:
First Time Setup (this includes if the end user has "factory reset" their device).

**OLED Display**
Change "myssid" to display a different message here to inform the end user that the device needs to be set up with their WiFi info from their WiFi router.

ie

"Connect to the SSID below to setup WiFi settings from your WiFi router !" 

_Image example enclosed_

[image]

skot, 2024-11-03:
I like this. We'll have to work on the wording a bit to get it to fit.

We can use SSID == MYSSID as the condition for this mode.

STSMiner1, 2024-11-03:
Here you go 

All the  .cvs    files for the board versions.... 

401 example below

''key,type,encoding,value
main,namespace,,
hostname,data,string,bitaxe
wifissid,data,string,WiFI SETUP REQUIRED
wifipass,data,string,password
stratumurl,data,string,public-pool.io
stratumport,data,u16,21496
stratumuser,data,string,bc1qnp980s5fpp8l94p5cvttmtdqy8rvrq74qly2yrfmzkdsntqzlc5qkc4rkq.bitaxe
stratumpass,data,string,x
fbstratumurl,data,string,solo.ckpool.org
fbstratumport,data,u16,3333
fbstratumuser,data,string,bc1qnp980s5fpp8l94p5cvttmtdqy8rvrq74qly2yrfmzkdsntqzlc5qkc4rkq.bitaxe
fbstratumpass,data,string,x
asicfrequency,data,u16,490
asicvoltage,data,u16,1166
asicmodel,data,string,BM1368
devicemodel,data,string,supra
boardversion,data,string,401
flipscreen,data,u16,1
invertfanpol,data,u16,1
autofanspeed,data,u16,1
fanspeed,data,u16,100
selftest,data,u16,1
overheat_mode,data,u16,0``

[image]

skot, 2024-11-03:
Wow, yes that's a nice simple way to do it!

I still think we might want to have a "Initial Setup Mode" screen to make this even easier for the first-timers

mutatrum, 2024-12-12:
This is fixed with #539.

https://github.com/skot/ESP-Miner/blob/ba6be3c3bbb827df471e22bf2ecc54112d3fd82c/main/screen.c#L117

STSMiner1, 2025-01-27:
See https://github.com/skot/ESP-Miner/pull/623

closed 2025-04-01 by mutatrum