bitaxeorg/ESP-Miner #454: JDHX BM1366

Peichan83, 2024-11-04:
Bonjour ,

Mon Bitaxe BM1366 est bloqué sur le wifi ou ne le reconnait plus.

Pouvez-vous m'aider?
[image]

STSMiner1, 2024-11-04:
@Peichan83,

Vérifiez les paramètres WiFi du Bitaxe - ils doivent correspondre à ceux du routeur WiFi.

Assurez-vous également que le Bitaxe n'est pas trop éloigné du routeur WiFi, car un signal faible peut également être à l'origine de ce problème.

chenyang2151, 2024-11-05:
is bm1397 not bm1366

STSMiner1, 2024-11-05:
@Peichan83 

Vous avez également le mauvais firmware / config installé sur cet appareil !

Puce / carte BM1397, pas BM1366 !

chenyang2151, 2024-11-05:
board version and aisc type select wrong ?

STSMiner1, 2024-11-05:
Looks like a Bitaxe Max, but I've never seen one without "Bitaxe" on the PCB, this looks like a clone of it.

chenyang2151, 2024-11-05:
it must be bm1397  use Max's config.csv

skot, 2024-11-05:
The manufacturer chose to violate the Bitaxe license and not publish their changes. I suggest contacting them for support. We cannot help you here.

closed 2024-11-05 by skot