bitaxeorg/ESP-Miner #508: feature: Add expected hashrate to API

b-rowan, 2024-11-22:
It would be nice to have the expected hashrate shown in the API directly to make it a bit easier for external tools to parse it.  AFAIU there is some calculation being done involving cores, frequency, etc, to find this value.

I am happy to PR this if someone can start me in the right direction as to where to source this.

jns-codeworks, 2024-11-24:
I think it is this line:

`this.expectedHashRate$ = this.info$.pipe(map(info => {
      return Math.floor(info.frequency * ((info.smallCoreCount * info.asicCount) / 1000))
    }))`

[source](https://github.com/skot/ESP-Miner/blob/master/main/http_server/axe-os/src/app/components/home/home.component.ts)

jpcomps, 2025-01-03:
have a first pass of it here, will clean up and submit pull request against this issue:

https://github.com/jpcomps/ESP-Miner/tree/add_expected_hr

@WantClue @skot are we ok with the precision here, or any recommendations if we want to push floats, assume we would just convert it to a string

mutatrum, 2025-06-11:
Fixed by #603

PR #603 (2025-05-14): feature: add expected hashrate to api
Fixes #508 

Adds expected HR in GH/s based on chip count, freq and small_core_count. There is a divide which could potentially have a small performance impact, but should be minimal/negligible as is only calculated when the endpoint is called.
closed 2025-06-11 by mutatrum