bitaxeorg/ESP-Miner #511: Master Branch - PR's / Updates that have been pushed / merged around 11/21/2022 to the Master branch are causing an issue with 205 board.

STSMiner1, 2024-11-23:
Release firmware v2.4.0 - 205 works fine.

Firmware built 11/21/22 onwards with recent updates / PR's that have been merged to the master branch are causing an issue with the 205, issue is not seen on Supra (401) or Gamma (601).

Logs enclosed (startup) for both firmware builds from the 205 board.

Releasefirmware-v2.4.0.txt

Firmware built 11/21/22 with latest updates
PreRelease-11.21.2022.txt  (v2.4.0-7-g4100402-dirty)

Screenshot of issue also enclosed.

[Releasefirmware-v2.4.0.txt](https://github.com/user-attachments/files/17878146/Releasefirmware-v2.4.0.txt)

[PreRelease-11.21.2022.txt](https://github.com/user-attachments/files/17878147/PreRelease-11.21.2022.txt)

[image]

STSMiner1, 2024-11-23:
Typo's in report (year), sorry about that.

STSMiner1, 2024-11-23:
Pulled the repo down fresh and re-built the firmware to rule that out, flashed both files to the Ultra 205, no change.

[image]

eandersson, 2024-11-23:
What happens if you change the frequency?

STSMiner1, 2024-11-23:
Nothing changes with how it performs, voltage readings and temps show wrong along with the temp.

ASIC Frequency updates on the Dashboard as does the Measured ASIC Voltage when changed in the Settings page.

eandersson, 2024-11-23:
Can you try to comment out this line to see if it makes a difference? might be some weird race condition
https://github.com/skot/ESP-Miner/blob/master/components/asic/serial.c#L51

mutatrum, 2024-11-24:
Can you zoom in on the actual commit by doing a `git bisect`?

skot, 2024-11-27:
Narrowing it down: problem does not happen in 17873069a2a5237f4e8910e3fd500018d9ea25c9 and happens in 41004029901a0127b626c6070495f9e0af7046d0

skot, 2024-11-27:
Well that is surprising; it's caused by the uart_wait_tx_done() on this line... https://github.com/skot/ESP-Miner/blob/29a543da76ba5fd19e07f43da87214613d67455a/components/asic/serial.c#L51

eandersson, 2024-11-27:
> Well that is surprising; it's caused by the uart_wait_tx_done() on this line...
> 
> https://github.com/skot/ESP-Miner/blob/29a543da76ba5fd19e07f43da87214613d67455a/components/asic/serial.c#L51
> 
> from PR #503
> 
> this smells like a sneaky timing problem.

Yep - this is what I was asking about in the firmware testing channel. It's a really sneaky problem, but has been difficult for me to troubleshoot as I unfortunately don't have any of the older devices.

eandersson, 2024-11-27:
btw during my testing I found that after changing the baud it left the device unable to communicate with the ASIC due to some sort of race condition on the Supra. This feels like the same issue, but different timing issue.

The original issue I was solving was basically if one CPU cycle was changed (e.g. removing a single log line before BAUD is setup) it would break the ASIC startup.

I wish we had some sort of sanity check after all setup is done, for my testing I just just the ASIC count check to verify that we could still communicated. 

```
11/23/2024 07:25:11 PM E (26772) MAIN: BAUD RATE set to 115201
11/23/2024 07:25:16 PM E (26772) MAIN: Number of ASIC: 1 (Setup successful)
```

WantClue, 2024-12-01:
fixed in #532