bitaxeorg/ESP-Miner #512: Chip-Temperature 2.3.0 -> 2.4.0

mark0st83, 2024-11-23:
I took the plunge yesterday and updated my Gamma 601 to version 2.4.0. Everything seems to have worked so far.

However, the fan is now running much faster than before. The chip temperature is showing about 10 degrees Celsius warmer than before.
Because of this, the fan is running about 2000 RPM faster.

The frequency and voltage remain the same, but this isn’t visible in the 2.3.0 screenshot.

What temperature or operating condition is considered correct now?

[image]
[image]

STSMiner1, 2024-11-23:
2.4.0 fixes an issue with temps being under reported in 2.3.0.

https://github.com/skot/ESP-Miner/pull/483

mark0st83, 2024-11-23:
Ok, so the 45 degrees reported in 2.3.0 was wrong? Ok thats a lot „inaccurat“ 😅

skot, 2024-11-23:
Gamma ASIC temperature measurement before v2.4.0 was incorrect. It could be too high or too low.

closed 2024-11-23 by skot