bitaxeorg/ESP-Miner #515: Feature: configurable BOOT button (e.g. turn display off/on)

terratec, 2024-11-25:
I would like to be able to turn the display on/off.

What do you think about the following implementation?
[image]

mutatrum, 2024-11-25:
What you could do is add a screensaver value, e.g. turn off after x seconds/minutes, or 0 to not turn it off.  If you press the boot button, screen turns on again for the same amount of time. That way it's a simpler configuration and we don't overload the button functionality.

I'm currently reworking button and screen behaviour to combine #510 into #434.

terratec, 2024-11-25:
Thanks, that sounds great. I will use it as practice.

terratec, 2024-11-26:
[image]

or maybe:
[image]

skot, 2024-11-28:
Man I wish we had _one more_ button on the Bitaxe to use for UI

mutatrum, 2024-11-29:
You can have different lengths of long press. So normal click, normal long
press (> 200ms or so) and hold (> 2 seconds).

mutatrum, 2025-11-05:
Fixed by #525