bitaxeorg/ESP-Miner #516: change selftest to allow the bitaxe to boot, even if tests fail

skot, 2024-11-26:
Currently if one of the selftest tests fail, the Bitaxe will be stuck in selftest mode forever, essentially bricking the Bitaxe for those who don't know how to flash the nvs.

I think that no matter if tests pass or fail we should set nvs selftest=0 and boot normally after the user presses the BOOT button.

Of course it should be very obvious on the OLED and in the logs that tests have failed, so that both users and factories are aware.

skot, 2024-11-26:
Also, once we have this functionality then it makes sense to use factory images (selftest=1) on the webflasher @WantClue

benjamin-wilson, 2024-11-26:
selftest=0 should only happen _**after**_ the user presses the BOOT button, so that it can be power cycled and re-tested.

skot, 2024-11-26:
I suppose we could get fancy and require a long press on BOOT to restart (and set selftest=0) if there is a failure?

skot, 2024-11-28:
This is addressed in PR #524

benjamin-wilson, 2024-12-22:
> selftest=0 should only happen _**after**_ the user presses the BOOT button, so that it can be power cycled and re-tested.

When the self test passes, it should not self test again though, sorry if this was miscommunicated. This issue arose when trying to flash 2.4.2.
@skot

skot, 2024-12-22:
> > selftest=0 should only happen _**after**_ the user presses the BOOT button, so that it can be power cycled and re-tested.
> 
> 
> 
> When the self test passes, it should not self test again though, sorry if this was miscommunicated. This issue arose when trying to flash 2.4.2.
> 
> @skot 

I thought maybe you'd want to test the boot button as that final test. Fine with me either way.

closed 2025-01-06 by WantClue