bitaxeorg/ESP-Miner #519: Feedback after update not appearing

tdb3, 2024-11-27:
Settings Page, Update Firmware and Update Website forms.

When uploading a new firmware file (e.g. ), the page shows a `Working...` overlay, then appears to show no feedback of the completion status of the update (i.e. success or failure). This was tested using Firefox and Chrome.

Looks like there is code for this, but the Toasts weren't observed. https://github.com/skot/ESP-Miner/blob/29a543da76ba5fd19e07f43da87214613d67455a/main/http_server/axe-os/src/app/components/settings/settings.component.ts#L186-L200

Running developer tools showed:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the `Settings` page
2. Click `Browse` in the `Update Website` form
3. Select `www.bin`

**Expected behavior**
Feedback is provided to the user.

**Hardware:**
 - Bitaxe HW version: Supra 401
 - ESP-Miner FW version: master (29a543da76ba5fd19e07f43da87214613d67455a)
 - Hash Frequency: N/A for issue
 - Voltage: N/A for issue
 - Pool URL, Port, User: N/A for issue

tdb3, 2024-11-27:
Looks like the intent of the existing code is to reload the page after `www.bin` update, which would be prudent (to force the user to use the new web UI rather than the client-side code remnant of the UI pre-update).

skot, 2024-11-30:
If we don't have to restart the ESP32 to reload the page, that works too!

tdb3, 2024-12-01:
#521 fixes the issue (and provides a nice percentage status while waiting!)

closed 2024-12-01 by tdb3