bitaxeorg/ESP-Miner #550: Add unit test execution to CI

tdb3, 2024-12-03:
Could be beneficial to add unit tests to CI, to prevent regressions from going unnoticed.

Wish list item for now.

There are probably a few ways to do this (e.g. abstracting hardware with mocks, having a self-hosted runner with physical esp32 devices attached, etc.).

https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/unit-tests.html

eandersson, 2024-12-03:
You can run tests using QEMU, there is even a Github Action that handles it for you, but it isn't maintained and only supports ESP-IDF 4.x.

eandersson, 2024-12-06:
Spent some time trying to get it to work, but kept running into crashes.
https://github.com/eandersson/ESP-Miner/actions/runs/12205882369/job/34057802320#step:6:55

eandersson, 2025-01-07:
I figured out the crash. QEMU does not support hardware accelerated SHA (CONFIG_MBEDTLS_HARDWARE_SHA) so that feature needs to be disabled, but that leads issues as the software implementation is slightly different.
https://github.com/eandersson/ESP-Miner/commit/43b5b4bb7eb163dbe8ef62bfd7e04d21a2b1f993#diff-382cd75d2d2b14dad2fd5104843e642afda81185c5c88a5870b64a5e7a6ffe91R226

tdb3, 2025-01-08:
Nice. Since no test is perfect, I think it seems reasonable to have layers of testing, each with their own limitations (and we clearly state those limitations). For example, the QEMU based CI for unit test execution might have to run mostly the same (but not identical) software, which enables finding some bugs.

I don't think it would be too crazy to use real ESP32s dev boards. They are cheap enough, and could enable testing code that we might not be able to test well with QEMU. It has a tiny bit of a Rube Goldberg machine vibe, but would be just one layer of testing.