bitaxeorg/ESP-Miner #559: Hide Realtime-Logs-Entries with diff lower than stratumDiff

seepv, 2024-12-06:
Would it be possible to hide all Realtime-Logs-Entries with a diff lower than stratumDiff ?

Instead of:
...
₿ (155208313) asic_result: Ver: 2F04C000 Nonce 74640230 diff 258.7 of 1000.
₿ (155212273) asic_result: Ver: 2E9EA000 Nonce 25D70288 diff 1731.0 of 1000.
₿ (155214383) asic_result: Ver: 2F9F0000 Nonce B98C02DA diff 316.9 of 1000.
₿ (155219083) asic_result: Ver: 2410E000 Nonce 68C102A6 diff 376.4 of 1000.
₿ (155221383) asic_result: Ver: 26AF4000 Nonce 1A120262 diff 2463.4 of 1000.
₿ (155223173) asic_result: Ver: 24CEE000 Nonce B35C01C8 diff 1051.1 of 1000.
₿ (155225263) asic_result: Ver: 259F0000 Nonce 233F02A4 diff 485.8 of 1000.
₿ (155225493) asic_result: Ver: 27AB2000 Nonce A3440316 diff 621.4 of 1000.
₿ (155225683) asic_result: Ver: 296BC000 Nonce 2E37018E diff 264.1 of 1000.
₿ (155229713) asic_result: Ver: 29AE4000 Nonce 161C00DE diff 663.3 of 1000.

A Button shows only these (>stratumDiff) :
...
₿ (155212273) asic_result: Ver: 2E9EA000 Nonce 25D70288 diff 1731.0 of 1000.
₿ (155221383) asic_result: Ver: 26AF4000 Nonce 1A120262 diff 2463.4 of 1000.
₿ (155223173) asic_result: Ver: 24CEE000 Nonce B35C01C8 diff 1051.1 of 1000.

mutatrum, 2024-12-08:
Some pools have really high stratum difficulty, so that would mean with this change there's hardly any logging.

seepv, 2024-12-11:
In the frontend it could look like this:

[image]