bitaxeorg/ESP-Miner #565: null values in API crash the device

w3irdrobot, 2024-12-08:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**

When updating settings through the API at `PATCH /api/system`, if any values are explicitly `null`, it seems to crash the entire device.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a Bitaxe.
2. Dance because you own a Bitaxe, and that's awesome.
3. Get device IP
4. Run the command `curl -v -X PATCH -d '{"wifiPass":null}' http://<DEVICE_IP>/api/system`
5. Notice that the device restarts itself (fans spin down, Uptime in web app resets)
6. Cry

**Expected behavior**

I would expect it to be ignored or an error be thrown to let me know I screwed up the request body.

**Screenshots & Photos**
If applicable, add AxeOS screenshots and/or photos of your Bitaxe to help explain your problem.

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma 601
 - Bitaxe HW vendor: Solo Satoshi
 - ESP-Miner FW version: v2.4.0 
 - Hash Frequency: 525
 - Voltage: 1160
 - Pool URL, Port, User: Braiins.....

**Additional context**

```
hostname
ssid
wifiPass
stratumURL
stratumUser
stratumPassword
fallbackStratumURL
fallbackStratumUser
fallbackStratumPassword
```

w3irdrobot, 2025-02-21:
anyone actually working on this yet? if not, i can work on it.

mutatrum, 2025-06-11:
Fixed by #748

PR #748 (2025-03-16): Improve null handling in API settings PATCH
- Adds a function for getting a string value from a JSON payload attribute and returns a boolean representing if it's a non-null value
- Updates value retrieval from the JSON settings object in the settings PATCH endpoint to correctly handle JSON `null` values as well as values that don't exist in the JSON payload

I am not a C developer. So there may be a more C-like way of doing this, but this does work. :smile: 

Closes #565
closed 2025-06-11 by mutatrum