bitaxeorg/ESP-Miner #581: AxeOS 2.4.1 incorrectly states restart is needed to change hostname

alltheseas, 2024-12-13:
_what happens_
1) change hostname
2) AxeOS displays restart required for changes to take effect
3) I observe that the hostname name is updated in Swarm view without a restart

[image]

_suggestion_

Consider removing the "restart needed" in case of hostname change

skot, 2024-12-13:
> Does the following logic work:
> 
> 1. if hostname only is changed, then do not prompt for a restart
> 2. if wifi info is changed, then prompt for a restart

Yes, I think that could work! We probably have this same situation on the settings tab too. Fan speed comes to mind. @mrv777 what do you think? How possible is this to have some logic in there to determine if a restart is actually needed?

kienthay, 2024-12-16:
Hello, new to this site and also bitcoin mining. I flashed the new version 2.4.1 for Gamma 601, however, after restart, during reinstalments with WIFI, the device did not let met connect... flashed it back to original version and I'm able to connect... what can I look into/do to make this a successful transition?

mrv777, 2024-12-22:
@skot Yeah, I can do that.  I'm not sure which list is shorter, but so we have a list of what changes REQUIRE a restart and/or what changes DON'T REQUIRE a restart?