bitaxeorg/ESP-Miner #588: For BM1370 miners which fall into a state of not mining, a soft restart of the device does not fix the problem.

ParkaMark, 2024-12-18:
**Describe the bug**
Compared to BM1366 and BM1368 miners, I am encountering a weird behaviour where BM1370 miners are falling into a state showing a 0 GH/s hash rate, pulling only 5 W of power. I have noticed this does happen from time to time (for reasons I do not fully know) but I have been able to fix the problem by calling the `/api/system/restart` API end point which sends the device for a reboot.

If this happens with my BM1366 or BM1368 miners, they normally come back and begin mining again after being restarted. However, with my BM1370 miners, this does not fix the problem and they continue to not mine. The only way to resolve this issue is a total power cycle of the miner.

I'm speculating that there is some fundamental hardware difference and thus behaviour with the BM1370 which the existing code is not accounting for so it would be great to get some feedback and traction on this bug.

**Expected behavior**
I expect a POST to `/api/system/restart` to restart the miner and restore mining on BM1370 devices, but mining does not restart after the device has restarted. The same behaviour is observed when using a browser and clicking on the restart button within the miner's website.

**Screenshots & Photos**
Here is a dump of the output of my management BASH script showing 3 miners that are currently not mining, and 2 that are.

```
http://bm1370-0  	25 °C	 5 W	 50 %	   0 GH/s	 601 @ v2.4.2	BM1370 @ 550 MHz	1.00 cV @ 0.00 aV	ota_1
http://bm1370-1  	56 °C	16 W	 50 %	 810 GH/s	 601 @ v2.4.2	BM1370 @ 550 MHz	1.00 cV @ 0.99 aV	ota_1
http://bm1370-2  	65 °C	17 W	 50 %	 782 GH/s	 601 @ v2.4.2	BM1370 @ 550 MHz	1.00 cV @ 0.97 aV	ota_1
http://bm1370-3  	28 °C	 5 W	 50 %	   0 GH/s	 601 @ v2.4.2	BM1370 @ 550 MHz	1.00 cV @ 0.00 aV	ota_1
http://bm1370-4  	26 °C	 5 W	 50 %	   0 GH/s	 601 @ v2.4.2	BM1370 @ 550 MHz	1.00 cV @ 0.00 aV	ota_1
```

**Hardware (please complete the following information):**
See above.

19201003080114, 2024-12-18:
I reported the same issue #544 

A warm restart doesn’t get it to work again. 
I had to power off completely, then power on again to make it work.

skot, 2024-12-18:
how often is your script polling the bitaxe API? do you see any change if you stop it?

skot, 2024-12-18:
If you could collect the USB logs from boot when this happens, that would be great

ParkaMark, 2024-12-18:
> how often is your script polling the bitaxe API? do you see any change if you stop it?

Hourly at most. I'd have to check and get back to you about stopping the polling but I don't see how hourly polling would be a problem.

Will try and grab USB logs but not something I've done before. I have managed to do a comparison just now on the logs in the UI and this is what that looks like, with filtering for `jobs` lines.

```
₿ (14251) create_jobs_task: Set chip version rolls 65535		  |	₿ (14286) create_jobs_task: Set chip version rolls 65535
₿ (14271) create_jobs_task: New Work Dequeued 0001047c			  |	₿ (14306) create_jobs_task: New Work Dequeued 000104aa
₿ (14281) create_jobs_task: Job processed and queued: 0001047c		  |	₿ (14316) create_jobs_task: Job processed and queued: 000104aa
									  >	₿ (21736) create_jobs_task: New Work Dequeued 000104ab
									  >	₿ (21736) create_jobs_task: Job processed and queued: 000104ab
									  >	₿ (29836) create_jobs_task: New Work Dequeued 000104ac
									  >	₿ (29836) create_jobs_task: Job processed and queued: 000104ac
									  >	₿ (31746) create_jobs_task: New Work Dequeued 000104ad
									  >	₿ (31746) create_jobs_task: Job processed and queued: 000104ad
									  >	₿ (41746) create_jobs_task: New Work Dequeued 000104ae
									  >	₿ (41746) create_jobs_task: Job processed and queued: 000104ae
```
So on a warm restart (left), it does a single job processed and queued and then there is nothing further. After power cycling (right), it does that and then continues, and it's mining again. Also, I did see traffic going to and from the mining pool while it was not mining, so not sure what that signifies (if anything).

Let me know if there's anything else I can help with investigating/troubleshooting. Thanks!

19201003080114, 2024-12-18:
I suspect there is a bad state kept in TPS546D24ARVFR (U2), as in the bad state it only draws 5W. 
It perhaps cleared during a power cycle and it works again.

skot, 2024-12-18:
> I suspect there is a bad state kept in TPS546D24ARVFR (U2), as in the bad state it only draws 5W but everything else works - just not mining. It perhaps cleared during a power cycle and it works again.

that's an interesting idea. Hopefully OP can get USB logs from boot in both situations where it working and not.

asheiner, 2025-01-09:
I'm having the same issue with my BM1370.

However, it never starts/started mining. Regardless of soft or cold booting my bitaxe.
And I'm never getting any message about a processed job.
Only repeating 
"stratum_task: rx:" and "create_jobs_task: New Work Dequeued "

`₿ (12383) bm1370Module: Setting Frequency to 131.25MHz (131.25)
₿ (12483) bm1370Module: Setting Frequency to 137.50MHz (137.50)
₿ (12593) bm1370Module: Setting Frequency to 143.75MHz (143.75)
₿ (12693) bm1370Module: Setting Frequency to 150.00MHz (150.00)
₿ (12803) bm1370Module: Setting Frequency to 156.25MHz (156.25)
₿ (12903) bm1370Module: Setting Frequency to 162.50MHz (162.50)
₿ (13013) bm1370Module: Setting Frequency to 168.75MHz (168.75)
₿ (13123) bm1370Module: Setting Frequency to 175.00MHz (175.00)
₿ (13223) bm1370Module: Setting Frequency to 181.25MHz (181.25)
₿ (13333) bm1370Module: Setting Frequency to 187.50MHz (187.50)
₿ (13433) bm1370Module: Setting Frequency to 193.75MHz (193.75)
₿ (13543) bm1370Module: Setting Frequency to 200.00MHz (200.00)
₿ (13643) bm1370Module: Setting Frequency to 206.25MHz (206.25)
₿ (13753) bm1370Module: Setting Frequency to 212.50MHz (212.50)
₿ (13853) bm1370Module: Setting Frequency to 218.75MHz (218.75)
₿ (13963) bm1370Module: Setting Frequency to 225.00MHz (225.00)
₿ (14063) bm1370Module: Setting Frequency to 231.25MHz (231.25)
₿ (14173) bm1370Module: Setting Frequency to 237.50MHz (237.50)
₿ (14273) bm1370Module: Setting Frequency to 243.75MHz (243.75)
₿ (14373) bm1370Module: Setting Frequency to 250.00MHz (250.00)
₿ (14473) bm1370Module: Setting Frequency to 256.25MHz (256.25)
₿ (14583) bm1370Module: Setting Frequency to 262.50MHz (262.50)
₿ (14693) bm1370Module: Setting Frequency to 268.75MHz (268.75)
₿ (14803) bm1370Module: Setting Frequency to 275.00MHz (275.00)
₿ (14903) bm1370Module: Setting Frequency to 281.25MHz (281.25)
₿ (15013) bm1370Module: Setting Frequency to 287.50MHz (287.50)
₿ (15113) bm1370Module: Setting Frequency to 293.75MHz (293.75)
₿ (15223) bm1370Module: Setting Frequency to 300.00MHz (300.00)
₿ (15323) bm1370Module: Setting Frequency to 306.25MHz (306.25)
₿ (15443) bm1370Module: Setting Frequency to 312.50MHz (312.50)
₿ (15543) bm1370Module: Setting Frequency to 318.75MHz (318.75)
₿ (15653) bm1370Module: Setting Frequency to 325.00MHz (325.00)
₿ (15783) bm1370Module: Setting Frequency to 331.25MHz (331.25)
₿ (15893) bm1370Module: Setting Frequency to 337.50MHz (337.50)
₿ (15993) bm1370Module: Setting Frequency to 343.75MHz (343.75)
₿ (16103) bm1370Module: Setting Frequency to 350.00MHz (350.00)
₿ (16203) bm1370Module: Setting Frequency to 356.25MHz (356.25)
₿ (16313) bm1370Module: Setting Frequency to 362.50MHz (362.50)
₿ (16413) bm1370Module: Setting Frequency to 368.75MHz (368.75)
₿ (16523) bm1370Module: Setting Frequency to 375.00MHz (375.00)
₿ (16623) bm1370Module: Setting Frequency to 381.25MHz (381.25)
₿ (16733) bm1370Module: Setting Frequency to 387.50MHz (387.50)
₿ (16833) bm1370Module: Setting Frequency to 393.75MHz (393.75)
₿ (16943) bm1370Module: Setting Frequency to 400.00MHz (400.00)
₿ (17043) bm1370Module: Setting Frequency to 406.25MHz (406.25)
₿ (17153) bm1370Module: Setting Frequency to 412.50MHz (412.50)
₿ (17253) bm1370Module: Setting Frequency to 418.75MHz (418.75)
₿ (17363) bm1370Module: Setting Frequency to 425.00MHz (425.00)
₿ (17463) bm1370Module: Setting Frequency to 431.25MHz (431.25)
₿ (17573) bm1370Module: Setting Frequency to 437.50MHz (437.50)
₿ (17673) bm1370Module: Setting Frequency to 443.75MHz (443.75)
₿ (17783) bm1370Module: Setting Frequency to 450.00MHz (450.00)
₿ (17883) bm1370Module: Setting Frequency to 456.25MHz (456.25)
₿ (17993) bm1370Module: Setting Frequency to 462.50MHz (462.50)
₿ (18093) bm1370Module: Setting Frequency to 468.75MHz (468.75)
₿ (18193) bm1370Module: Setting Frequency to 475.00MHz (475.00)
₿ (18303) bm1370Module: Setting Frequency to 481.25MHz (481.25)
₿ (18413) bm1370Module: Setting Frequency to 487.50MHz (487.50)
₿ (18513) bm1370Module: Setting Frequency to 493.75MHz (493.75)
₿ (18623) bm1370Module: Setting Frequency to 500.00MHz (500.00)
₿ (18723) bm1370Module: Setting Frequency to 506.25MHz (506.25)
₿ (18833) bm1370Module: Setting Frequency to 512.50MHz (512.50)
₿ (18933) bm1370Module: Setting Frequency to 518.75MHz (518.75)
[... 11 lines cut ...]
₿ (19243) stratum_task: Clean Jobs: clearing queue
₿ (19243) stratum_api: tx: {"id": 1, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}
₿ (19263) stratum_api: tx: {"id": 2, "method": "mining.subscribe", "params": ["bitaxe/BM1370/v2.4.5"]}
₿ (19273) stratum_api: tx: {"id": 3, "method": "mining.authorize", "params": ["bc1qj8fgvvqg72cu8dtv20yp6jft7pq4y6dsvqqrqv.bitaxe", "anything123"]}
₿ (19283) stratum_api: tx: {"id": 4, "method": "mining.suggest_difficulty", "params": [1000]}
₿ (19293) stratum_task: rx: {"result":{"version-rolling":true,"version-rolling.mask":"1fffe000"},"id":1,"error":null}
₿ (19303) stratum_api: Set version mask: 1fffe000
₿ (19303) stratum_task: Set version mask: 1fffe000
₿ (19323) stratum_task: rx: {"result":[[["mining.notify","67111564"]],"fffd1067",8],"id":2,"error":null}
₿ (19333) stratum_api: extranonce_str: fffd1067
₿ (19333) stratum_api: extranonce_2_len: 8
₿ (19343) stratum_task: rx: {"params":[1000],"id":null,"method":"mining.set_difficulty"}
₿ (19353) stratum_task: Set stratum difficulty: 1000
[... cut ...] 33919a08bcae2cdc98c6745","19a4bd55c6e42f99c880a01a5edc4c4492980c5f02ffc971e4cdb0d397460404","aec38e97e5fcdeedd693d56e2bdb27931db3c1adf159031ec8da621bcf8d98ce","6b65608958ff371464014349768151ff5ede9c1fbf5c8c95a27af6de81678131","bc8224c728f60cf216fbb9c6f801af817558231db61a83b744c6f8277c3f3505","b3bd699e729f4e3d7181cc8641af91ce2d87ed9c68d33b8326f7e24679b859c4"],"20000000","1702905c","677f91cb",false],"id":null,"method":"mining.notify"}
₿ (55253) create_jobs_task: New Work Dequeued 670b6aaa0003f08d`

jeroenubbink, 2025-01-09:
> I'm having the same issue with my BM1370.
> 
> However, it never starts/started mining. Regardless of soft or cold booting my bitaxe. And I'm never getting any message about a processed job. Only repeating "stratum_task: rx:" and "create_jobs_task: New Work Dequeued "

Yes same issue with my BM1370. I've had it powered down for more than an hour.

skot, 2025-01-09:
Yes, I think we have an issue here. My guess is still some setting cached in the voltage regulator.

It would be very helpful if someone could get USB logs FROM BOOT of the bitaxe starting in this bad state.

jeroenubbink, 2025-01-09:
I'm sure i could manage but unfortunately my psu seems broken now as well
so i'll have to get that replaced first.

matlen67, 2025-01-09:
I currently have a Gamma here from a user in the Bitcoin forum for verification/repair. His Gamma stopped working after a fan upgrade and he asked me if I could take a look at it because I posted in the forum that I occasionally do something with electronics as a hobby. The Gamma has a consumption of 5 watts and does not start with the Minig. When measuring, I noticed that pin 5 of U6 (MPT1824T-0802E) has 1.2V instead of 0.8V. I cannot judge whether this is the cause. I have ordered new MPTs and will try to replace them. Enclosed is the log of the system start.

Systemlog:

=~=~=~=~=~=~=~=~=~=~=~= KiTTY log 2025.01.09 21:15:44 =~=~=~=~=~=~=~=~=~=~=~=
I (138837) http_server: Restarting System because of API Request
I (139837) wifi:state: run -> init (0x0)
I (139847) wifi:pm stop, total sleep time: 0 us / 127240756 us

I (139847) wifi:<ba-del>idx:1, tid:0
I (139847) wifi:<ba-del>idx:0, tid:6
I (139847) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
I (139857) wifi_station: Could not connect to 'MBTC' [rssi -49]: reason 8
I (139907) wifi:flush txq
I (139907) wifi:stop sw txq
I (139907) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375eb5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (26) boot: ESP-IDF v5.4 2nd stage bootloader
I (26) boot: compile time Jan  7 2025 15:36:30
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (29) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 16MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00400000
I (77) boot:  3 www              Unknown data     01 82 00410000 00300000
I (84) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (90) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (97) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (103) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (110) boot: End of partition table
I (113) boot: Defaulting to factory image
I (117) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=2d0e4h (184548) map
I (157) esp_image: segment 1: paddr=0003d10c vaddr=3fc9c800 size=02f0ch ( 12044) load
I (160) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=d2d9ch (863644) map
I (313) esp_image: segment 3: paddr=00112dc4 vaddr=3fc9f70c size=02730h ( 10032) load
I (315) esp_image: segment 4: paddr=001154fc vaddr=40374000 size=18764h (100196) load
I (338) esp_image: segment 5: paddr=0012dc68 vaddr=600fe100 size=0001ch (    28) load
I (348) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (359) octal_psram: vendor id    : 0x0d (AP)
I (359) octal_psram: dev id       : 0x02 (generation 3)
I (360) octal_psram: density      : 0x03 (64 Mbit)
I (364) octal_psram: good-die     : 0x01 (Pass)
I (369) octal_psram: Latency      : 0x01 (Fixed)
I (375) octal_psram: VCC          : 0x01 (3V)
I (380) octal_psram: SRF          : 0x01 (Fast Refresh)
I (386) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (391) octal_psram: BurstLen     : 0x01 (32 Byte)
I (397) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (403) octal_psram: DriveStrength: 0x00 (1/1)
I (409) MSPI Timing: PSRAM timing tuning index: 5
I (414) esp_psram: Found 8MB PSRAM device
I (418) esp_psram: Speed: 80MHz
[... 370 lines cut ...]
I (14123) stratum_api: Resetting stratum uid
I (14123) stratum_task: Clean Jobs: clearing queue
I (14123) stratum_api: tx: {"id": 1, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}
I (14143) stratum_api: tx: {"id": 2, "method": "mining.subscribe", "params": ["bitaxe/BM1370/v2.4.5"]}
I (14153) stratum_api: tx: {"id": 3, "method": "mining.authorize", "params": ["bc1q4hrn7up6w9mwagjpwssgsqafuykgvjqzyhhpje", "x"]}
I (14163) stratum_api: tx: {"id": 4, "method": "mining.suggest_difficulty", "params": [1000]}
I (14173) stratum_task: rx: {"result":{"version-rolling":true,"version-rolling.mask":"1fffe000"},"id":1,"error":null}
I (14183) stratum_api: Set version mask: 1fffe000
I (14183) stratum_task: Set version mask: 1fffe000
I (14193) stratum_task: rx: {"result":[[["mining.notify","67ac3648"]],"6b86a967",8],"id":2,"error":null}
I (14203) stratum_api: extranonce_str: 6b86a967
I (14213) stratum_api: extranonce_2_len: 8
I (14213) stratum_task: rx: {"params":[10000],"id":null,"method":"mining.set_difficulty"}
I (14223) stratum_task: Set stratum difficulty: 10000
I (14233) stratum_task: rx: {"params":["67484e7d0001ec67","e55e85d8085e010006d7b65b6ac7f7984c5593c50001db360000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff3503c2670d0004652e8067040f8b281e0c","0a636b706f6f6c112f736f6c6f2e636b706f6f6c2e6f72672fffffffff033b48621200000000160014adc73f703a7176eea24174208803a9e12c864802ec0b60000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed67b3 [... cut ...] b36e6dc51c81b6e21d3a73ca","bd4e24e6a5b9b477da1b169ef28fe4e3cb0533d13e4005a0c439dcbe623af941","389368ecba7061c19cac1246f36c834016974e36d4df1fc0b81a051819e0aa15","87f8116028ae41e4afbf533207f01ef9fe791dddc8b1faf15895b8e73fb03f11","3f33101db40b77ca8a17a92026ab58eb054e5ee8fdda03a61a89002a75d3b80a","166b5d8108d83efb71ead66e508795387bbc5e0be565e695c1b53db5af13a680"],"20000000","1702905c","67802e83",false],"id":null,"method":"mining.notify"}
I (15903) create_jobs_task: New Work Dequeued 67484e7d0001ec68

matlen67, 2025-01-10:
> > I currently have a Gamma here from a user in the Bitcoin forum for verification/repair. His Gamma stopped working after a fan upgrade and he asked me if I could take a look at it because I posted in the forum that I occasionally do something with electronics as a hobby. The Gamma has a consumption of 5 watts and does not start with the Minig. When measuring, I noticed that pin 5 of U6 (MPT1824T-0802E) has 1.2V instead of 0.8V. I cannot judge whether this is the cause. I have ordered new MPTs and will try to replace them. Enclosed is the log of the system start, it may help.
> 
> Do you think this could maybe also brick your PSU in some way?

My power supplies are ok, my Gamma works perfectly with them.

skot, 2025-01-11:
> I currently have a Gamma here from a user in the Bitcoin forum for verification/repair. His Gamma stopped working after a fan upgrade and he asked me if I could take a look at it because I posted in [...]

Thank you for the detailed log. This is really strange about the ourput voltage of the 0.8V LDO.. please let us know what your find out.

skot, 2025-01-11:
Another one. https://x.com/StackAddict90/status/1878163312354922757

skot, 2025-01-12:
I compared the log from @matlen67 with one from a working Gamma here. No differences that I can tell. Still digging.

MyOwn2C, 2025-01-29:
One more case 
https://x.com/anonameus0/status/1884334945885462961?s=46&t=161hsAiA9KLPWjeFpSKW8g

MyOwn2C, 2025-01-31:
Another case:
https://x.com/dockcapital/status/1885207797593260499?s=46&t=161hsAiA9KLPWjeFpSKW8g

bestel74, 2025-02-03:
Same issue here, when the problem is present, here is the logs output : 

`₿ (32683094) bm1370Module: Job ID: 58, Core: 36/3, Ver: 1BBE6000

₿ (32683094) bm1370Module: Invalid job nonce found, 0x58

₿ (32684094) bm1370Module: Job ID: 58, Core: 93/5, Ver: 07F2A000

₿ (32684094) bm1370Module: Invalid job nonce found, 0x58

₿ (32684724) bm1370Module: Job ID: 58, Core: 45/12, Ver: 0F958000

₿ (32684724) bm1370Module: Invalid job nonce found, 0x58`

In my case, it could be an overclocking issue, running at 1.32v measured and 1100MHz. (Temperature are fine tho)

gustavobrossi, 2025-02-08:
I've got same issue for a couple of weeks now, if I recall, it all started when I bumped to v2.5.1. Previously reconnecting the PSU would bring it back up but now has gone completely dead. Not even flashing older firmware versions works. With 2.5.1 I've got no useful logs but factory resetting with 2.4.5 brings interesting results from the self test.

It is unclear to me if there is still a fix for the TPS546 as flashing other versions doesn't bring it back to life.

`Script started on Fri Feb  7 23:10:14 2025
command: screen /dev/tty.usbmodem14101 115200
[?1049h[!p[?3;4l[4l>[4l[?1h=[0m(B[1;60r[H[2J[H[2JI (158) esp_image: segment 1: paddr=0003d10c vaddr=3fc9c800 size[32mI (2485) bm1370Module: 1 chip(s) detected on the chain, expected 1[39m
tx: [55 AA 51 09 00 A4 90 00 FF FF 1C]
tx: [55 AA 51 09 00 A8 00 07 00 00 03]
tx: [55 AA 51 09 00 18 F0 00 C1 00 04]
tx: [55 AA 53 05 00 00 03]
tx: [55 AA 40 05 00 00 1C]
tx: [55 AA 51 09 00 3C 80 00 8B 00 12]
tx: [55 AA 51 09 00 3C 80 00 80 0C 11]
[32mI (2495) bm1370Module: Setting ASIC difficulty mask to 255[39m
tx: [55 AA 51 09 00 14 00 00 00 FF 08]
tx: [55 AA 51 09 00 58 00 01 11 11 0D]
tx: [55 AA 41 09 00 A8 00 07 01 F0 15]
tx: [55 AA 41 09 00 18 F0 00 C1 00 0C]
tx: [55 AA 41 09 00 3C 80 00 8B 00 1A]
tx: [55 AA 41 09 00 3C 80 00 80 0C 19]
tx: [55 AA 41 09 00 3C 80 00 82 AA 05]
tx: [55 AA 51 09 00 B9 00 00 44 80 0D]
tx: [55 AA 51 09 00 54 00 00 00 02 18]
tx: [55 AA 51 09 00 B9 00 00 44 80 0D]
tx: [55 AA 51 09 00 3C 80 00 8D EE 1B]
[32mI (2555) bm1370Module: Ramping up frequency from 56.25 MHz to 525.00 MHz with step 6.25 MHz[39m
tx: [55 AA 51 09 00 08 40 A2 02 55 0F]
[32mI (2565) bm1370Module: Setting Frequency to 56.25MHz (56.25)[39m
tx: [55 AA 51 09 00 08 40 AF 02 64 08]
[32mI (2575) bm1370Module: Setting Frequency to 62.50MHz (62.50)[39m
tx: [55 AA 51 09 00 08 40 A5 02 54 08]
[32mI (2675) bm1370Module: Setting Frequency to 68.75MHz (68.75)[39m
tx: [55 AA 51 09 00 08 40 A8 02 63 11]
[32mI (2775) bm1370Module: Setting Frequency to 75.00MHz (75.00)[39m
tx: [55 AA 51 09 00 08 40 B6 02 63 0C]
[32mI (2875) bm1370Module: Setting Frequency to 81.25MHz (81.25)[39m
tx: [55 AA 51 09 00 08 40 A8 02 53 1A]
[32mI (2975) bm1370Module: Setting Frequency to 87.50MHz (87.50)[39m
tx: [55 AA 51 09 00 08 40 B4 02 53 12]
[32mI (3075) bm1370Module: Setting Frequency to 93.75MHz (93.75)[39m
tx: [55 AA 51 09 00 08 40 A8 02 62 14]
[32mI (3175) bm1370Module: Setting Frequency to 100.00MHz (100.00)[39m
tx: [55 AA 51 09 00 08 40 AA 02 43 15]
[32mI (3275) bm1370Module: Setting Frequency to 106.25MHz (106.25)[39m
tx: [55 AA 51 09 00 08 40 A2 02 52 14]
[32mI (3375) bm1370Module: Setting Frequency to 112.50MHz (112.50)[39m
tx: [55 AA 51 09 00 08 40 AB 02 52 12]
[32mI (3475) bm1370Module: Setting Frequency to 118.75MHz (118.75)[39m
tx: [55 AA 51 09 00 08 40 B4 02 52 17]
[32mI (3575) bm1370Module: Setting Frequency to 125.00MHz (125.00)[39m
tx: [55 AA 51 09 00 08 40 BD 02 52 11]
[32mI (3675) bm1370Module: Setting Frequency to 131.25MHz (131.25)[39m
tx: [55 AA 51 09 00 08 40 A5 02 42 0C]
[32mI (3775) bm1370Module: Setting Frequency to 137.50MHz (137.50)[39m
tx: [55 AA 51 09 00 08 40 A1 02 61 1D]
[32mI (3875) bm1370Module: Setting Frequency to 143.75MHz (143.75)[39m
tx: [55 AA 51 09 00 08 40 A8 02 61 1B]
[32mI (3975) bm1370Module: Setting Frequency to 150.00MHz (150.00)[39m
tx: [55 AA 51 09 00 08 40 AF 02 61 19]
[32mI (4075) bm1370Module: Setting Frequency to 156.25MHz (156.25)[39m
tx: [55 AA 51 09 00 08 40 B6 02 61 06]
[32mI (4175) bm1370Module: Setting Frequency to 162.50MHz (162.50)[39m
tx: [55 AA 51 09 00 08 40 A2 02 51 1B]
[32mI (4275) bm1370Module: Setting Frequency to 168.75MHz (168.75)[39m
tx: [55 AA 51 09 00 08 40 A8 02 51 10]
[32mI (4375) bm1370Module: Setting Frequency to 175.00MHz (175.00)[39m
tx: [55 AA 51 09 00 08 40 AE 02 51 0A]
[32mI (4475) bm1370Module: Setting Frequency to 181.25MHz (181.25)[39m
tx: [55 AA 51 09 00 08 40 B4 02 51 18]
[... 343 lines cut ...]
[32mI (13975) self_test: Nonce 2322138226 Nonce difficulty 13.96450421696506261071135668316856.[39m
[32mI (13975) self_test: 906.773251 Gh/s  , duration 2.879816[39m
[32mI (14015) bm1370Module: Job ID: 18, Core: 65/6, Ver: 03A6C000[39m
[32mI (14015) self_test: Nonce 2619670914 Nonce difficulty 10.85713927935993083906396350357682.[39m
[32mI (14015) self_test: 905.228598 Gh/s  , duration 2.922687[39m
[32mI (14025) bm1370Module: Job ID: 18, Core: 48/0, Ver: 03B20000[39m
[32mI (14035) self_test: Nonce 2845966688 Nonce difficulty 16.64329129234445048268753453157842.[39m
[32mI (14035) self_test: 911.596058 Gh/s  , duration 2.939964[39m
[32mI (14045) bm1370Module: Job ID: 18, Core: 73/6, Ver: 03EAC000[39m
[32mI (14055) self_test: Nonce 2365194642 Nonce difficulty 8.79606344137725493226298567606136.[39m
[32mI (14065) self_test: 916.746455 Gh/s  , duration 2.960927[39m
[32mI (14085) bm1370Module: Job ID: 18, Core: 21/2, Ver: 04884000[39m
[32mI (14085) self_test: Nonce 882049834 Nonce difficulty 28.09970398970457949872070457786322.[39m
[32mI (14085) self_test: 917.737553 Gh/s  , duration 2.995169[39m
[32mI (14155) bm1370Module: Job ID: 18, Core: 74/5, Ver: 0568A000[39m
[32mI (14155) self_test: Nonce 1432879508 Nonce difficulty 29.46805603652905602984901634044945.[39m
[32mI (14165) self_test: 907.331843 Gh/s  , duration 3.067388[39m
[32mI (14165) self_test: Hashrate: 907.331843[39m
[32mI (14175) self_test: Voltage: 1117[39m
[32mI (14175) self_test: Power: 18.105835, Voltage: 1.148438, Current 15.765625[39m
[31mE (14185) self_test: TPS546 Power Draw Failed, target 11.00[39m
[32mI (14195) self_test: SELF TESTS FAIL -- Press RESET to continue[39m

Script done on Fri Feb  7 23:10:43 2025
`

MaSe-Time, 2025-03-02:
I'm having the same problem here with a bitaxe gamma 601, trying multiple different clocks/voltages but after a random amount of time it just stops mining and only a full power off and on again will clear it. Is this something that could be fixed with a firmware update?

skot, 2025-03-02:
> I'm having the same problem here with a bitaxe gamma 601, trying multiple different clocks/voltages but after a random amount of time it just stops mining and only a full power off and on again will clear it. Is this something that could be fixed with a firmware update?

Yes, I think this can be fixed with a firmware update. I'm going to try and get this into the next release, v2.6.0

skot, 2025-03-02:
> I've got same issue for a couple of weeks now, if I recall, it all started when I bumped to v2.5.1. For a couple of days, reconnecting the PSU would bring it back up but now has gone completely dead. Not even flashing older firmware versions works. With 2.5.1 I've got no useful logs but factory resetting with 2.4.5 brings interesting results from the self test.
> 
> It is unclear to me if there is still a fix for the TPS546 as flashing other versions doesn't bring it back to life.
> 
> ```
> Script started on Fri Feb  7 23:10:14 2025
> command: screen /dev/tty.usbmodem14101 115200
> I (158) esp_image: segment 1: paddr=0003d10c vaddr=3fc9c800 size
> 
> (2485) bm1370Module: 1 chip(s) detected on the chain, expected 1
> ~snip
> (14175) self_test: Voltage: 1117
> (14175) self_test: Power: 18.105835, Voltage: 1.148438, Current 15.765625
> E (14185) self_test: TPS546 Power Draw Failed, target 11.00
> (14195) self_test: SELF TESTS FAIL -- Press RESET to continue
> 
> Script done on Fri Feb  7 23:10:43 2025
> ```

This seems to be a different issue. leave you bitaxe unplugged for a few minutes and then try the selftest again.

MaSe-Time, 2025-03-02:
> Yes, I think this can be fixed with a firmware update. I'm going to try and get this into the next release, v2.6.0

Brilliant, it's my first miner and was thinking of sending it back 👀

Any idea when v2.6.0 maybe released? 🙈

MaSe-Time, 2025-03-03:
> Yes, I think this can be fixed with a firmware update. I'm going to try and get this into the next release, v2.6.0

Another suggestion. I find hashing is higher when the ASIC temp is around the 60c mark. If I set the fan to auto (I have a Argon Thrml installed) it doesn't spin the fan down slow enough so I end up with a very cold ASIC. Is there a way to add in to the GUI under settings a "target temp" that the fan will auto adjust to try and meet? It would need to be able to use the full range of 0 RPM to 100 RPM though as when mine stops hashing with the issue above, the slowest the fan spins is 35% even when the ASIC shows -1 degrees.

Anyway, just a suggestion.

MyOwn2C, 2025-03-05:
> Yes, I think this can be fixed with a firmware update. I'm going to try and get this into the next release, v2.6.0

I am seeing this issue reported almost every day now, by users from different sources: TG, X, FB, etc
Only fix that seems to work is power down and power back up again.

skot, 2025-03-05:
> I am seeing this issue reported almost every day now, by users from different sources: TG, X, FB, etc Only fix that seems to work is power down and power back up again.

I have a fix that I _think_ should fix this, but it's tough to know for sure since I can't reproduce the bug myself. If you or anyone you hear about can reliably reproduce this, and can also handle flashing beta firmware, please lmk on OSMU

MaSe-Time, 2025-03-05:
> I have a fix that I _think_ should fix this, but it's tough to know for sure since I can't reproduce the bug myself. If you or anyone you hear about can reliably reproduce this, and can also handle [...]

I'm happy to try, just bought a second one to try as it is so annoying constantly stopping.

skot, 2025-03-05:
> I'm happy to try, just bought a second one to try as it is so annoying constantly stopping.

If my theory is correct, this is caused by a power supply fault. So the question then is; why are you having so many power supply faults? Can you give me more details about your setup?

- How often does this happen?
- What Bitaxe hardware version do you have?
- Where did you get it?
- What power supply do you have?
- Anything else special about your setup?

MaSe-Time, 2025-03-05:
I have this gamma from Amazon
https://amzn.eu/d/a5t9QLl

Bought this 100w psu as well
https://amzn.eu/d/4xAR5fq 

But to be honest despite there being no dips in voltage I had thought a better quality PSU may be in order so am waiting for a 200w meanwell to be delivered. 

I had it running for about 4 or 5 days gradually increasing the clocks no problem but then all of a sudden it would just crash. Bringing it back to stock seems to run OK but it has crashed on a couple of occasions after 6 or 8 hours at close to stock clocks and voltages. 

I've tried the latest beta firmware with no luck and flashing back to 2.5.1 is just the same. 

Is there a possibility of damaging the hardware overclocking? My temps have always been good as you can see my cooling is pretty good, I keep it at 60c on the ascic and the vrm hasn't ever gone above 74

Attached is my setup

[image]
[image]

Thank you for your help

MaSe-Time, 2025-03-05:
> If my theory is correct, this is caused by a power supply fault. So the question then is; why are you having so many power supply faults? Can you give me more details about your setup?
>
> * How often [...]

FYI my new one just did it, same set up as above had clocks at 925mhz 1380mV. Only thing I can think is it's the power supply. Mean We'll 200w coming in a couple of days

skot, 2025-03-05:
> FYI my new one just did it, same set up as above had clocks at 925mhz 1380mV. Only thing I can think is it's the power supply. Mean We'll 200w coming in a couple of days

Can you try running it for a while on defaults and see if it happens?

MaSe-Time, 2025-03-05:
Sure, I'll run them both overnight olt default

MaSe-Time, 2025-03-06:
Ok, New one stopped over night and dropped to 5w. I restarted this morning and its stopped again, this time it looks like its hashing but it isn't.

[image]

skot, 2025-03-06:
This seems like a different issue. You should contact the seller about this.

MaSe-Time, 2025-03-06:
> This seems like a different issue. You should contact the seller about this.

OK, I'll return that one. It still did the original fault though whereby it stopped hashing and dropped to 5w much like my first one does

MaSe-Time, 2025-03-08:
OK, here's a new one. Both stop hashing at exactly the same time 🤷

Running on the new mean well lrs-200-5

[image]

MaSe-Time, 2025-03-08:
Guessing it may be pool related, swapped to eusolo.ckpool.org

MaSe-Time, 2025-03-08:
> Guessing it may be pool related, swapped to eusolo.ckpool.org

OK, it's not the pool, 10 mins in it did it again. This is a nightmare 🙄

benjamin-wilson, 2025-03-08:
> > Guessing it may be pool related, swapped to eusolo.ckpool.org
> 
> OK, it's not the pool, 10 mins in it did it again. This is a nightmare 🙄

Unstable ASIC, return to defaults

MaSe-Time, 2025-03-08:
> > > Guessing it may be pool related, swapped to eusolo.ckpool.org
> > 
> > 
> > OK, it's not the pool, 10 mins in it did it again. This is a nightmare 🙄
> 
> Unstable ASIC, return to defaults

Both at exactly the same time both clocked differently?

skot, 2025-03-08:
Are these attached to the same power supply? Same power strip?

MaSe-Time, 2025-03-08:
> Are these attached to the same power supply? Same power strip?

Both connected to my new mean well lrs-200-5. Outputting 5.4v. 

I've now bought a third bitaxe as well to see how they behave, all connected to the same psu and will return one  or all if they're not stable. 

First unit cooled with argon THRML 60 plus copper heatsinks. Clocked at 800mhz 1310mV requested, 1.28v reported

Next two units heatsinks removed, re pasted with hydronaut and put back in their cases. All in the same cool outside cupboard

Both clocked at 675mhz 1280mV requested, 1.25v reported

[image]

MaSe-Time, 2025-03-09:
1 and 2 both dropped to 5w after 1.2 hours and 2.4 hours respectively. They then after an hour or so both stopped hashing but looked like they were hashing so put clocks and voltages to default and warm restarted them. 

They then both stopped hashing over night again but were able to be warm restarted again this morning.

Unit 3 continues to hash at 675 1280mV.

I pinged them all and all were getting 30 to 40ms with the occasional high ping in the low 100ms

Do you have any idea when 2.6.0 maybe released so I can try it before returning at least two of these?

Thanks for listening

MaSe-Time, 2025-03-09:
Also, if at all possible could the fan parameters be changed to target 60c and expand the fan % range down to 20%? Could make the target temp user selectable even

skot, 2025-03-09:
> 1 and 2 both dropped to 5w after 1.2 hours and 2.4 hours respectively. They then after an hour or so both stopped hashing but looked like they were hashing so put clocks and voltages to default and warm restarted them. 
> 
> They then both stopped hashing over night again but were able to be warm restarted again this morning.
> 
> Unit 3 continues to hash at 675 1280mV.
> 
> Do you have any idea when 2.6.0 maybe released so I can try it before returning at least two of these?
> 
> Thanks for listening 

We are targeting the end of March for the v2.6.0 release. I don't think this is going to solve your problem though.

This upcoming release will still stop mining as a result of a power fault like this. It will show a banner in AxeOS and require a manual reset, just like overtemp mode.

I think in your case the Bitaxe voltage regulator (TPS546) is shutting down because of a fault situation occurring on your Bitaxe or from your PSU. There isn't much esp-miner can do about that besides just turn off and let you know something is wrong.

MaSe-Time, 2025-03-09:
Thanks Skot, I appreciate your insights. 

Looks like I'll be returning two of these. The third one seems stable at the moment and I've just put an Argon THRML on it and copper heatsinks.

Do you think in these instances it could reboot itself as the whole point of these after initially playing around with clocks etc is to leave them going with little to no intervention.

Could be a warning come up once logged in to say something along the lines of "system auto rebooted x times due to x, check psu/clocks etc" 

Could set it so it can auto reboot only 4/5 times in a 24 hour period after which it defaults to an error state until you log in? 

Just for the record this occurred with two separate power supplies, a cheap 100w amazon one and a better quality meanwell. 

Again, thanks for your insights, much appreciated

skot, 2025-03-09:
The bug we are working on fixing here is this no-hashing mode that a **warm restart does not fix**.  I'd like to stay focused on that here so we can get it resolved.

I created https://github.com/skot/ESP-Miner/issues/752 to track the isssue that a warm restart _does_ fix.

MaSe-Time, 2025-03-09:
> > Both connected to my new mean well lrs-200-5. Outputting 5.4v.
> 
> The Bitaxe max input voltage is 5.5V. Maybe you are getting a VIN overvoltage fault? Can you try turning this down so that all connected Bitaxe are at or below 5V and see if the problem persists?

I missed this skot, I've dialed it down so the two that keep crashing are showing 5v now. The third one is currently on a thinner gauge wire so is 0.2v lower due to droop. Will update 🙈

MaSe-Time, 2025-03-10:
Ok, my last gasp attempt was to move all three units out of my front cupboard and next to an additional router in my rear outbuilding. Tested pings and all sub 5ms.... 12 hours gone by and no drop outs at all no drop to 5w and stop hashing which requires a hard reset and no stop hashing but still pulling the same power and soft reset starts it again.

Moral of the story, if you think you have adequate wifi signal, you haven't! Move them closer to the router!!

Ill post this in the other thread you created as well skoot.

MavenCH, 2025-03-10:
Evening everyone

I have had the same problem since last week.
It worked great until it dropped to 0.
I tried restarting via the website and also disconnecting the miner from the power for a few minutes without success.
Changing the pool didn't change anything and neither did placing the miner next to an access point.

Then I tried to reload the firmware but also without success.
Downgrading to e.g. 2.5.0 has also had no effect, nor has giving it more power or less.

I get the same messages in the log and sometimes the following:
`₿ (314651) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[128]}
₿ (314661) stratum_task: Set stratum difficulty: 128
₿ (314661) stratum_task: rx: {"id":null,"method":"mining.notify","params":["d3ace4","5dd991299afabe1cc5097059607430ed5f242e6c00016d4b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703a4890d5075626c69632d506f6f6c","ffffffff02102cc8120000000017a914b46e629336cac9252929b9808cb1ea7d4af14f59870000000000000000266a24aa21a9ed9a03889efaed8f47ae98bd6be0d84cb74d478623bb9ca2eae09f654ef250167500000000",["1fd29ed19514cd341321c814064fd244c2e70ed418b6ddb5a23fbc0982081ee6","8df0d6e9ff48ef975b883f2982616af4763f3ec0262675cdc68a0a92024e0f40","f7f0c15e1a6ac265228de4a4aa44a64cf58e8c7bffbcb9ff282149faa8b247ed","e02d42563a304b08529706165c9c5d8647b8b0fa141fff307ee2351a3a8df0a1","7adeb112a0a0654a83d332d9f4fdddbaa634b0356f14938ba6ad6a6015c6191f","fa2867ceca90aab4ca25952d3fa358ecb9efac1342715206653a37c8dfd08a50","1f330e4b1f873be30f86cb7ca95132c4723cafa0713caaf24087d68da569fdd3","14fc939db7093f67f1c1379de4d80568bafbf84c7d9252b47d59e9779063f696","17efc5f1f7eb9d5c45c39c0c340c7fa3260a5ebf218eb3d3c85afd4d9adedf72","1f9610c8953ccd75dd2d72f3179602468976408e403ed92dac232d14c0d97fcf","8e558a524b742659605b6f53409f70d605a98791269b1a4a14ac92557cee650d","e601cdaf9aa004c5a725091e6170e1d8a499d962d5ed7b64713f0fb6f028c2de"],"20000000","17028281","67cf2d5e",true]}
₿ (314771) stratum_task: Clean Jobs: clearing queue
₿ (314831) create_jobs_task: New Work Dequeued d3ace4
₿ (314831) ASIC_task: New pool difficulty 128
₿ (343861) stratum_task: rx: {"id":null,"method":"mining.notify","params":["d3b8a9","5dd991299afabe1cc5097059607430ed5f242e6c00016d4b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703a4890d5075626c69632d506f6f6c","ffffffff02aa8ecd120000000017a914b46e629336cac9252929b9808cb1ea7d4af14f59870000000000000000266a24aa21a9ed5d0f4f3f14e644ebb8312811580931971db38be0dcb57c7cf537dda23e63ddb100000000",["1fd29ed19514cd341321c814064fd244c2e70ed418b6ddb5a23fbc0982081ee6","8df0d6e9ff48ef975b883f2982616af4763f3ec0262675cdc68a0a92024e0f40","b80435c480acc519edc6084a24b58d3c661d6fad86a34028c8df4b4016c0c534","359512eb308b18df351462332b5b709824f81bd216719abf755b65af8968c805","183bf93d099bb630bc3cbb68986e78bdb865d1e47c5e60827a40315fb1d20287","80efdd9ec78db947762b366f9354f3552a4142e0d8271c6a0d52e93ba27c2534","84042d7b1af047236a387a8b51726a1909bf5f4422302649ce96cac3bc2ec124","ccea336f16ba2192fae06e94ed4844500fd8b947d1ad82042d5f4edaa757ed04","22b3742e18cf255546650772800c29cf3f6ecb1374401ff869277eae1f151972","fd95688cab3c91261528912dcaaf4e80f210ac1f0431820ec0ca18e48068ab37","0e34112582af694334c74d32d5c32b33568a6702e3b934732f71dfaf66fa4a0a","58a6464d0ed6732cacafe03221e690010e2eb5fc7c3ed1a937c28e87ed0a84e0"],"20000000","17028281","67cf2d9a",false]}
₿ (344041) create_jobs_task: New Work Dequeued d3b8a9
₿ (374651) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[16]}
₿ (374651) stratum_task: Set stratum difficulty: 16
₿ (374661) stratum_task: rx: {"id":null,"method":"mining.notify","params":["d3b8fe","5dd991299afabe1cc5097059607430ed5f242e6c00016d4b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703a4890d5075626c69632d506f6f6c","ffffffff02aa8ecd120000000017a914b46e629336cac9252929b9808cb1ea7d4af14f59870000000000000000266a24aa21a9ed5d0f4f3f14e644ebb8312811580931971db38be0dcb57c7cf537dda23e63ddb100000000",["1fd29ed19514cd341321c814064fd244c2e70ed418b6ddb5a23fbc0982081ee6","8df0d6e9ff48ef975b883f2982616af4763f3ec0262675cdc68a0a92024e0f40","b80435c480acc519edc6084a24b58d3c661d6fad86a34028c8df4b4016c0c534","359512eb308b18df351462332b5b709824f81bd216719abf755b65af8968c805","183bf93d099bb630bc3cbb68986e78bdb865d1e47c5e60827a40315fb1d20287","80efdd9ec78db947762b366f9354f3552a4142e0d8271c6a0d52e93ba27c2534","84042d7b1af047236a387a8b51726a1909bf5f4422302649ce96cac3bc2ec124","ccea336f16ba2192fae06e94ed4844500fd8b947d1ad82042d5f4edaa757ed04","22b3742e18cf255546650772800c29cf3f6ecb1374401ff869277eae1f151972","fd95688cab3c91261528912dcaaf4e80f210ac1f0431820ec0ca18e48068ab37","0e34112582af694334c74d32d5c32b33568a6702e3b934732f71dfaf66fa4a0a","58a6464d0ed6732cacafe03221e690010e2eb5fc7c3ed1a937c28e87ed0a84e0"],"20000000","17028281","67cf2d9a",true]}
₿ (374771) stratum_task: Clean Jobs: clearing queue
₿ (374841) create_jobs_task: New Work Dequeued d3b8fe
₿ (374841) ASIC_task: New pool difficulty 16
₿ (403951) stratum_task: rx: {"id":null,"method":"mining.notify","params":["d3c4c5","5dd991299afabe1cc5097059607430ed5f242e6c00016d4b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703a4890d5075626c69632d506f6f6c","ffffffff023d3cd1120000000017a914b46e629336cac9252929b9808cb1ea7d4af14f59870000000000000000266a24aa21a9ed54c25fe0480159cb6594ae8cdf0db8c5f56e433c2488a2899b7fa069ee1910fd0000000 [... cut ...] 343","11ef82dad61734c3a7a49fbb7776ed0c34972aef389942536b1ad2814e444631","d027d1a805f68d0d25a78b798d98f5ff6aeb099926f31f137c16582077dfddcf","9033c674ce177ea9295146bad05defaad66bb01f9437f5c15cbf52f18b46e4d4","66aca04039e97ffd378d2f0dd9f40d6565a255bac944cc1cb63f3b73998065a6"],"20000000","17028281","67cf2e12",false]}
₿ (464381) create_jobs_task: New Work Dequeued d3d0d4`

I hope a solution will be found soon. Because I have only had this miner since 28.2.25

Many thanks
Kind regards from Switzerland

skot, 2025-03-10:
> > > Evening everyone
> > > I have had the same problem since last week. It worked great until it dropped to 0. I tried restarting via the website and also disconnecting the miner from the power for a few minutes without success. Changing the pool didn't change anything and neither did placing the miner next to an access point.
> > > Then I tried to reload the firmware but also without success. Downgrading to e.g. 2.5.0 has also had no effect, nor has giving it more power or less.
> > > I get the same messages in the log and sometimes the following: ...
> > > I hope a solution will be found soon. Because I have only had this miner since 28.2.25
> > > Many thanks Kind regards from Switzerland
> > 
> > 
> > What pool are you on? eusolo.ckoool.org was the lowest ping for me in the UK
> 
> Hello MaSe-Time
> 
> I have always been on public-pool.io:21496 and it has worked. But I have also tried to use eusolo.ckpool.org:3333, but without success.
> 
> The settings are set to default. In other words: 525 1150
> 
> I have also just tried to update to v2.6.0b8 but no change here either.
> 
> I have just tried to ping both but without success. Is there a special address that can be pinged?
> 
> Greetings
> 
> `--- web.public-pool.io ping statistics --- 27 packets transmitted, 27 received, 0% packet loss, time 67ms rtt min/avg/max/mdev = 14.178/16.162/18.555/1.178 ms `

MavenCH, 2025-03-10:
> > Evening everyone
> > I have had the same problem since last week. It worked great until it dropped to 0. I tried restarting via the website and also disconnecting the miner from the power for a few minutes without success. Changing the pool didn't change anything and neither did placing the miner next to an access point.
> > Then I tried to reload the firmware but also without success. Downgrading to e.g. 2.5.0 has also had no effect, nor has giving it more power or less.
> > I hope a solution will be found soon. Because I have only had this miner since 28.2.25
> 
> Where did you get this Bitaxe? What Power Supply do you have?
> 
> I'd like to collect some info on what power supplies people who see this error have.

Hello skot

Thank you for your request.
I got it from here: https://miningwholesale.eu/product/bitaxe-gamma/

Also here a picture of the miner with the power supply and the dashboard:
https://photos.app.goo.gl/rsoQ2PonckK8iAxr9

I have just seen from the picture that I have had the miner since 3.3.25 and put it into operation the day after.

I hope this helps.
For further questions I am very happy to answer :)

Greetings

Erkman84, 2025-03-16:
Same issue, downgrade does not help: restarting device and plug power off-on again does not solve the issue. It is a software releated issue... please fix it.

skot, 2025-03-16:
> Same issue: restarting (cold/warm) device and plug power off-on again does not solve the issue. It is a software releated issue... please fix it.
> 
> Update: downgrade to v2.0.3 seems to help. (Better as non-working miner)

We are going to have better indication that a Power fault has occurred in v2.6.0; to be released at the end of this month. You can load the latest pre-release beta now if you want to test it.

MavenCH, 2025-03-16:
> We are going to have better indication that a Power fault has occurred in v2.6.0; to be released at the end of this month. You can load the latest pre-release beta now if you want to test it.

I have already tried this, but unfortunately it has not changed anything.
I'll try to supply the miner via USB C tomorrow to rule out the power supply unit.

Since I have tried to downgrade to 2.0.3, I can no longer reach the miner. I hope I haven't made anything worse now...

MaSe-Time, 2025-03-16:
> I have already tried this, but unfortunately it has not changed anything. I'll try to supply the miner via USB C tomorrow to rule out the power supply unit.
>
> Since I have tried to downgrade to [...]

I have three that were all doing variations of this. Moved closer to my router and not had one drop out since 🤷

bonifacio123, 2025-03-18:
Running 2.6.0b10 - having no issues until I moved the gamma which was about 15 feet away from the wifi router to about 1 foot away and it manifested the non-mining issue. I just kept rebooting it until it just started working again. Using a 50w power supply. I didn't grab the logs but it was getting work from the pool but not mining.

cniweb, 2025-03-19:
I have the same problem with a Bitaxe 600, no Firmware Version´s work for me.

skot, 2025-03-24:
This should be "fixed" in #780 ... meaning that power faults are now shown in AxeOS, and assuming the fault cause is gone, a warm restart should clear.

ParkaMark, 2025-04-02:
Just an update from me on this. Have updated to 2.6.1. It's great that it now reports a power fault. I've since swapped the power supplies on my miners around. On one of my bulk orders from BitcoinMerch, they shipped units with some PSUs which are bricks with attachable power cables. All the other ones they've sent me on other orders where bricks where there is no attachable power cable, so the power socket pins are built straight into the unit that you plug in.

Anyway, after swapping them around, and putting the attachable power cable variants on my lowest power draw miners, and using the no attachable power cable variants on the 1370s, things are now looking way more stable.

Thank you for everyone involved and for fixing and addressing this bug. It seems that not all PSUs are equal.

[5 more replies]