bitaxeorg/ESP-Miner #598: [Feature Request] Redisplay Firmware Version on Device's Screen

crimsonleaf363, 2024-12-26:
# Description
I'm requesting a feature from previous versions be brought back. I would like the firmware version to be displayed on the device's screen.

# Benefits
- Convenient firmware version tracking
- Simplified troubleshooting

# Proposed Implementations
- The firmware version is displayed on the device's screen.
- The firmware version is displayed in Axe OS.

skot, 2024-12-26:
Why do users need to be constantly reminded of the firmware version at a glance? It's not changing all the often..

19201003080114, 2024-12-27:
Software versions is already available in Swarm, Settings, Logs and API.

crimsonleaf363, 2025-01-12:
> Why do users need to be constantly reminded of the firmware version at a glance? It's not changing all the often..

I use it to confirm firmware updates without having to log into AxeOS.

crimsonleaf363, 2025-01-12:
> Software versions is already available in Swarm, Settings, Logs and API.

This feature request is about being able to see the firmware version without having to check AxeOS.

skot, 2025-01-12:
The firmware version is available in AxeOS. I think the Bitaxe screen is already too clutter to show anything more.

closed 2025-01-12 by skot