bitaxeorg/ESP-Miner #618: cannot update to 2.4.3 firmware / website - bitaxe gamma 601 crashing #213

filipponeri, 2025-01-05:
Hi, I cannot update to the 2.4.3 firmware / website on a bitaxe gamma 601. 
The upload of the two files keeps freezing the web interface. After restart the onboard display does not turn on. I had to downgrade back to 2.4.2.

matlen67, 2025-01-06:
I had the same problem today when updating from 2.4.3 -> 2.4.4. The updates stopped between 30% and 99% of the time. After several update attempts, I entered an invalid address in the stratum host (also fallback) so that it would not start mining after a restart (host= abc.de). After that the update worked.

filipponeri, 2025-01-06:
same problem with the 2.4.4 release

filipponeri, 2025-01-06:
adding (irrelevant) features to the code base results in the code becoming unmanageable and in the appearance of weird crashes. Please slim down the code base, we want a home base miner to do experiments on it. So we need to understand and rely on the code base. Otherwise we would have bought a private code based miner.

skot, 2025-01-06:
> adding (irrelevant) features to the code base results in the code becoming unmanageable and in the appearance of weird crashes. Please slim down the code base, we want a home base miner to do experiments on it. So we need to understand and rely on the code base. Otherwise we would have bought a private code closed source based miner.

Feel free to fork esp-miner and slim it down however you like.

WantClue, 2025-01-07:
resolved with v2.4.5

closed 2025-01-07 by WantClue

filipponeri, 2025-01-07:
SOLVED with v2.4.5. It works.