bitaxeorg/ESP-Miner #633: [FEATURE REQUEST] Add "avg hash rate" and "avg Efficiency"

ovik88, 2025-01-09:
Hey, would it be possible to add Avg values for hash rate and efficiency?
[image]

skot, 2025-01-09:
We could do this now with just a running average over the current session (ie how long the browser window has been open)

There are future plans to store some historical performance data onboard, which would make this much more useful.

mrv777, 2025-01-10:
If we add avg hash to the card, I'm wondering if it should be removed from the chart to clean the chart up.  Maybe people like seeing the history of the avg hash there, but not sure if its needed

skot, 2025-01-11:
> If we add avg hash to the card, I'm wondering if it should be removed from the chart to clean the chart up. Maybe people like seeing the history of the avg hash there, but not sure if its needed

I think that makes sense.. I find myself wanting to know the average hashrate, and sometimes it's tough to read off the chart.

closed 2025-02-16 by WantClue