bitaxeorg/ESP-Miner #639: Access error after applying firmware v2.5.0b1

Master0ne, 2025-01-10:
After updating firmware to v2.5.0b1 trying to update website gives the following error:

[image]

The Swarm page then also can not access data of that Bitaxe anymore:

[image]

A downgrade of firmware is then also not possible anymore using the Bitaxe's website:

[image]

alexnb7, 2025-01-11:
Did you bought it on Aliexpress? I did it and i received a paper saying not to do any update on it ( they don’t explain why, just say that it will not work anymore)

skot, 2025-01-11:
There was a bug in 2.5.0b1. You should give the latest 2.5.0b6 a try. You'll need to ressurect your bitaxe over USB with the web flasher -> https://flasher.bitaxe.org

alexnb7, 2025-01-11:
Yeah , try it on https://flasher.bitronics.store/ and let us know if it works

alexnb7, 2025-01-11:
Yeah, let us know if it works👍🏻

closed 2025-01-11 by skot

Master0ne, 2025-01-12:
Unfortunately the same problem with v2.5.0b6. Had to go back to v2.4.5 again using the bitaxetool.