bitaxeorg/ESP-Miner #645: Updating by site nerdaxe

alexnb7, 2025-01-11:
[image]

this is the version that is on my nerdaxe.

could be great to know if it is possible to update by a site

MyOwn2C, 2025-01-11:
Unless you bought from OSMU sellers, you take your own risk by upgrading it using an official software.
Nerdaxe will lost its display function if you used the wrong software.

alexnb7, 2025-01-11:
I did it! By oficial software, everything great on this latest version🙌🏻

closed 2025-01-11 by WantClue

skot, 2025-01-11:
this is not the repo for nerdaxe firmware

alexnb7, 2025-01-11:
> Unless you bought from OSMU sellers, you take your own risk by upgrading it using an official software. Nerdaxe will lost its display function if you used the wrong software. Latest official software for Nerdaxe is "v2.1.9-36-gaecb8af-dirty"

Sure! It’s work!