bitaxeorg/ESP-Miner #668: Swarm error: "could not save" when entering IP address

Ben-in-Chicago, 2025-01-18:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
Since updating to v2.5.0, an error is presented when attempting to add devices to the Swarm menu.

**To Reproduce**
On the Swarm menu of any BitAxe, enter the IP address of another BitAxe. The error "Could not save" is shown and the device is not added to the Swarm.

**Expected behavior**
Device is added.

**Screenshots & Photos**

[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma, Supra and Ultra
 - Bitaxe HW vendor: 
 - ESP-Miner FW version: 2.5.0
 - Hash Frequency:
 - Voltage:
 - Pool URL, Port, User:

**Additional context**
I was running v2.3.0 on the Supra and Ultra and v2.4.0 on the Gamma when Swarm previously worked.

STSMiner1, 2025-01-19:
When updating the firmware you need to flash both the esp-miner.bin & www.bin files to the device otherwise your going to have problems.

Ben-in-Chicago, 2025-01-19:
Thanks! I just need to figure out how to change the display back to the prior view. The bright red isn't desired.

closed 2025-01-19 by Ben-in-Chicago