bitaxeorg/ESP-Miner #674: OTA firmware update fails if a websocket is open

dustinb, 2025-01-26:
**Describe the bug**
OTA firmware update fails if there is a web socket open to the Bitaxe
 
**To Reproduce**
Steps to reproduce the behavior:
1. Open 2 browser tabs to the same Bitaxe
2. Open and start logs in one of them
3. Update the firmware in the other tab
4. Firmware update hangs and eventually shows an error
5. Close the tab with the logging
6. Firmware update completes

**Expected behavior**
Firmware update completes successfully

[image]

[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: [201, 401, 601]
 - ESP-Miner FW version: [various]

eandersson, 2025-02-18:
I tried to reproduce this multiple times but never ran into the described issue. Both `esp-miner.bin` and `www.bin` succeeded without issues while viewing realtime logs at the same time.

eandersson, 2025-02-19:
I was finally able to reproduce this issue. It's basically possible to exhaust the stack when combining certain actions e.g. firmware upgrade + streaming logs.

PR #723 (2025-02-20): Fix stackoverflow when upgrading #674
This fixes #674 - It's possible to cause a stack overflow if upgrading the firmware and streaming logs at the same time.

```
I (37849) esp_image: segment 0: paddr=00b10020 vaddr=3c0e0020 size=2e738h (190264) map

***ERROR*** A stack overflow in task httpd has been detected.

Backtrace: 0x40375fc5:0x3fce0bc0 0x4037f151:0x3fce0be0 0x40380086:0x3fce0c00 0x403819d7:0x3fce0c80 0x4038014c:0x3fce0ca0 0x40380142:0x40377ae1 |<-CORRUPTED
--- 0x40375fc5: panic_enable_cache at /opt/esp/idf/components/esp_system/port/panic_handler.c:233
0x4037f151: prvCopyDataToQueue  [... cut]
closed 2025-02-20 by WantClue