bitaxeorg/ESP-Miner #680: "-1 oC" asic temperature value

alltheseas, 2025-01-30:
This is displayed while I am troubleshooting the unhappy path of overheat mode. 

I am guessing this is an incorrect measurement (I am not in the North Pole, or the Canadian tundra), or some failsafe. 

Consider displaying the actual temperature value, or a high temperature/overheat warning instead. 

[image]

MyOwn2C, 2025-01-30:
-1C is what it will show before it’s initialized

alltheseas, 2025-01-30:
> -1C is what it will show before it’s initialized

does "initialized" mean "starting to hash"?

MyOwn2C, 2025-01-30:
> > -1C is what it will show before it’s initialized
> 
> does "initialized" mean "starting to hash"?

From power up. 
After a few seconds it will read normal temperature.

skot, 2025-02-02:
I agree showing a real (albeit cold AF) temperature is confusing. Maybe we can show "--" instead.

closed 2025-02-12 by skot