bitaxeorg/ESP-Miner #691: 0 shares

ovik88, 2025-02-03:
Is this ok? HAving 0 shares yet, geting solved blocks?

[image]

MyOwn2C, 2025-02-03:
If it stays like this for more than 10 mins, then it’s not working. 
If you have nonce but no shares, you likely used a pool with high diff.

ovik88, 2025-02-03:
i didnt change pool and it was working just fine before.. but i was playing with voltage and freq. so mby i messed up something. I hade to restart it few times and now im hassing again, but mby it was issue on public-pool.io im mining , dunno

MyOwn2C, 2025-02-03:
Pubic pool was down over the weekend

skot, 2025-02-04:
Hmmm. This is a strange one. You said you were changing settings -- it would seem like the share counter got reset without resetting the best diff somehow.

ovik88, 2025-02-04:
> Hmmm. This is a strange one. You said you were changing settings -- it would seem like the share counter got reset without resetting the best diff somehow.

nop, i think it was just opposite. Share counter got stucked at 0 while "best difficulties -  since system boot" were updating everytime i found new high

skot, 2025-02-04:
> nop, i think it was just opposite. Share counter got stucked at 0 while "best difficulties - since system boot" were updating everytime i found new high

Ok, then it's prolly what @MyOwn2C said. Can you restart your Bitaxe and try this again now that public-pool is back up?

closed 2025-02-15 by WantClue