bitaxeorg/ESP-Miner #699: make AxeOS gauge scale adjustable for different Bitaxe models

skot, 2025-02-12:
[image]

Currently it looks like the scaling for the AxeOS scales are hardcoded. This is tough when we need to support different Bitaxe models with (some drastic) differences in readings.

Ideally AxeOS would update the gauge scales from a call to a function in `/main/power/power.c`

Also, it would be nice if the Input Voltage nominal marker was always in the middle of the gauge.

skot, 2025-02-18:
this is to support https://github.com/skot/ESP-Miner/issues/718

w3irdrobot, 2025-02-21:
yeah I can look into this. if the values we get through the info endpoint have been hooked up, then this should be pretty simple.

w3irdrobot, 2025-03-07:
@skot perhaps i'm missing it, but do we have a way of obtaining some of these board-specific constants right now? maybe i'm just missing it in the code, but I only see, in `power.c` for example, how we get the current value, not the maximum allowed value based on the board.

skot, 2025-03-07:
> @skot perhaps i'm missing it, but do we have a way of obtaining some of these board-specific constants right now? maybe i'm just missing it in the code, but I only see, in `power.c` for example, how we get the current value, not the maximum allowed value based on the board. 

You're right we don't have functions to return those board specific constants yet. I think power.c is probably the right place for them.

If you're working on the front end components can you just pretend they exist and then I'll add them?

WantClue, 2025-03-27:
#793 closes this

closed 2025-03-27 by WantClue