bitaxeorg/ESP-Miner #751: stuck while uploading in recovery mode

AnPel-K1, 2025-03-09:
Hi, 

after updating www.bin to 2.51 my LV06 booted into recovery mode. "Please upload www.bin to recover AxeOS" After choosing www.bin und push button "uploading" nothing happened any more. Several upload tries are unsuccessful. What to do?

AnPel-K1, 2025-03-09:
AI
Analyzing network data
Okay, let's break down this network request and try to understand why it's taking a significant amount of time.

Purpose of the Request:

URL: http://192.168.10.124/api/system/OTAWWW
This URL suggests an API endpoint on a local network (private IP address).
/api/system/OTAWWW implies that it's interacting with some system functionality, potentially related to Over-The-Air (OTA) updates or some other system-level operation designated by "OTAWWW."
Method: The method isn't explicitly stated, but Content-Type: application/octet-stream and Content-Length: <redacted> in the request headers strongly suggest this is a POST request sending binary data.
Content-Type: application/octet-stream indicates that arbitrary binary data is being sent, which could be a firmware image, configuration file, or some other type of raw data for the OTAWW operation.
Origin/Referer: The request is made from the same IP, this indicates a local network interaction.
Accept-Language: The language headers indicate that the browser is set to prefer German.
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/133.0.0.0 Safari/537.36 identifies the client as a Chrome browser on a macOS system.
Analysis of Timing Information:

Status Code: 0
A status code of 0 is highly unusual. It suggests the request likely failed before a proper HTTP response could be received. It's possible the request got blocked on the client side or that the server does not respond.
Queued at: -110,170,892 μs
Started at: -110,169,083 μs
The Queued/started phase took: 1,809 μs, or 1.809 ms
Connection start (stalled): 1.81 ms
This is the time the request spent stalled, while waiting for the connection to begin.
Duration: 38.91 seconds
This is the total time the request was active. A duration of nearly 39 seconds for a local network request is extremely long.
Summary The Request gets stuck and was never fullfilled.
Potential Issues and Reasons for the Long Duration:

Server Unreachable/Not Responding: The status code 0 and the long duration, strongly suggest that the server at 192.168.10.124 either isn't running the service needed to handle this request, isn't responding, or is having serious performance issues.
Firewall/Network Configuration: A firewall on the local machine, on the device at 192.168.10.124, or on the network itself might be blocking the request or preventing a response.
Overload/Resource Exhaustion: The target device might be overloaded, perhaps busy with other tasks, or it might be running out of memory or processing power, causing it to be unable to handle the request promptly.
Large Request Payload: Although the data size is redacted, if a significant binary data amount is uploaded it might take a while.
Bug in Server-Side Code: There might be a bug in the /api/system/OTAWWW endpoint code that is causing it to hang or perform very slowly.
No Server side handling: The web server might not provide a proper handler for this request.
Debugging Steps:

Verify Server Availability:
Check if the service that handles /api/system/OTAWWW is running on 192.168.10.124.
Try accessing other basic endpoints on 192.168.10.124 to see if the server is generally responsive.
Check Firewalls: Temporarily disable firewalls on the client and server (if possible) to rule out firewall interference.
Monitor Server Resources: If you have access to the device at 192.168.10.124, check its CPU, memory, and network usage while making the request.
Check for server errors: The server should return a error message in case of problems.
Simplify Request: Try sending a minimal request (a very small binary payload, or even an empty request if the endpoint allows it) to see if the problem is payload-size related.
Try with other Browser: Try to send the request from a different Browser.
Check client side code: If possible review the code that generated this request.
Check server side code: If possible review the server side code that should handle this request.
In summary: This request is exhibiting a major problem, likely on the server side, and the status code 0 further indicates a failed request. The extended duration of 38.91 seconds is a strong indicator of serious issues. The next step should be to confirm the availability and responsiveness of the target server at 192.168.10.124.

skot, 2025-03-09:
LuckyMiners are not supported.

closed 2025-03-09 by skot