bitaxeorg/ESP-Miner #759: FEATURE REQUEST:  Histogram for Total Hash Rate on the swarm page

jtsmith0101, 2025-03-11:
I like graphs.  Seeing a histogram on the swarm page similar to what is displayed on the individual worker pages would be cool, IMO.

Cheers,

James

dskvr, 2025-04-13:
IMO should be done in a separate interface. The AxeOS runs off a ESP32 and so has an extremely conservative space budget. Additionally, long term statistics aren't stored on device for similar reasons, there is no historical data.

There is an API (which is how the swarm functionality works) and so running an interface to handle this off-device is entirely possible; on your computer, start9 or umbrel for instance.

g1ass1, 2025-05-21:
I have implemented a very basic MQTT broker on my bit axe 601 (and I mean basic, it exports difficulty presently, but it works - this was done just to see if it was possible)- this connects to my home assistant instance which runs MQTT and I intend to use the data from there.

The api is good, but mqtt is better for RealTIME or nearly Realtime data  - probably abit overkill on this use case though?