bitaxeorg/ESP-Miner #761: Deal with RGB LEDs on bitaxeMax models

skot, 2025-03-11:
The old bitaxeMax models have two RGB LEDs. We never really used them, but some recent firmware change caused the `X` LED to be on a steady yellowish green. At the very least we should turn off LEDs on models that have them.

[image]

skot, 2025-03-11:
[image]

skot, 2025-03-11:
Bonus challenge is to make the LEDs do something cool, like flash when a share is submitted (ala the bitHalo). But make sure to properly abstract this HW-specific code in esp-miner

WantClue, 2025-03-13:
I think we removed a big portion of the code associated to the LEDs so it would be helpful to look into old commits

skot, 2025-03-18:
Well.... it looks like I used the same pins for the right LED (LEDX) that are used for Octal SPI (GPIO35, 36, 37) Which is used for PSRAM on the ESP32-S3-WROOM-1 module. It looks like the few bitaxeMax users out there are going to have to deal with the LED being on, or stay with with way old firmware from before we were using PSRAM.

[image]

[image]

closed 2025-03-18 by skot