bitaxeorg/ESP-Miner #773: Webserver Crash When Using AxeOS via Mobile Browser

LisBerndt, 2025-03-14:
When operating the AxeOS via a mobile browser (in this case, Yandex), the web server crashes after a few actions (viewing logs, opening the dashboard, change settings etc.). The API gets unreachable and in some cases, the device even becomes unreachable via ping. As a test, I used the OS exclusively through a desktop browser (Chrome) for an entire day—without any issues.

Bitaxe Gamma BM1370
AxeOS 2.5.1

Uefi1, 2025-03-17:
Yeah, I also noticed that the web interface sometimes disconnects. I have to turn off the ASIC, wait for it to cool down, and then turn it back on.

NotAnotherHelloWorld, 2025-03-17:
I experience similar issues. Web frontend worked fine for a week without issues, but as soon as I started collecting stats using curl on API endpoint, the webserver becomes unresponsive after a few hours. Ping works fine even when web is down. WiFi deauth and letting it reconnect solves this, without having to reboot the device. I'll try to see if I can gather something useful from the logs when I have the time, maybe it's not closing sessions correctly.

Uefi1, 2025-03-17:
> I experience similar issues. Web frontend worked fine for a week without issues, but as soon as I started collecting stats using curl on API endpoint, the webserver becomes unresponsive after a few [...]

I have some suspicions that the ESP32 chip is overheating because it's located near the heatsink, where the fan blows out hot air. Some of that hot air might be hitting the ESP32, which could be why the web interface stops responding.

LisBerndt, 2025-03-17:
> > I experience similar issues. Web frontend worked fine for a week without issues, but as soon as I started collecting stats using curl on API endpoint, the webserver becomes unresponsive after a few hours. Ping works fine even when web is down. WiFi deauth and letting it reconnect solves this, without having to reboot the device. I'll try to see if I can gather something useful from the logs when I have the time, maybe it's not closing sessions correctly.
> 
> I have some suspicions that the ESP32 chip is overheating because it's located near the heatsink, where the fan blows out hot air. Some of that hot air might be hitting the ESP32, which could be why the web interface stops responding.
> 
> BitAxe is probably poorly designed—it would have been better to solder the ESP32 chip on the back of the board rather than placing it next to the processor's cooling system.

I don't think it's a heating issue. This behavior first appeared about two hours in under default settings in a cool room, and as I mentioned, only when using a mobile user. In the last few days, while overclocking the device (with manual settings and additional cooling) and using only a desktop browser, the system has been running stably.

skot, 2025-03-17:
> I experience similar issues. Web frontend worked fine for a week without issues, but as soon as I started collecting stats using curl on API endpoint, the webserver becomes unresponsive after a few [...]

this is interesting. We were having problems running out of sockets but I thought #571 would have helped there. @WantClue do you know what release #571 made it into?

Uefi1, 2025-03-17:
> this is interesting. We were having problems running out of sockets but I thought [#571](https://github.com/bitaxeorg/ESP-Miner/pull/571) would have helped there. [...]

Skot, do you know when this issue occurs the most? Try refreshing the WiFi networks several times (about 5-7 times) using the magnifying glass icon to scan for signal and networks. After that, completely exit the interface and try logging in again!

By the way, there's another bug—the magnifying glass icon doesn't show all available networks, only 1-2, and sometimes none at all. It's likely that the WiFi network refresh time is too short.

Uefi1, 2025-03-19:
> this is interesting. We were having problems running out of sockets but I thought [#571](https://github.com/bitaxeorg/ESP-Miner/pull/571) would have helped there. [...]

Skot, I finally managed to flash the Lucky Miner LV06 with your esp-miner-factory-205 firmware! I really liked the firmware—thank you so much for your hard work and contribution!

hgs-sellis, 2025-08-06:
> IP changed? Does the pool receive the shares from the miner?

No, the IP has not changed. It seems to be related to how much you use the web UI. If I am careful not to check on them too much, they run longer without showing the problem.

Alex71btc, 2025-09-21:
still same issue with v2.10.0

hhaniel, 2025-12-10:
Same issue on v2.12.0 - My BitAxe is out in the garage so bit of a pain to get to all the time so I have it plugged into a smart socket so I can power cycle it from remote.