bitaxeorg/ESP-Miner #776: Change "Invert Fan Polarity" text

skot, 2025-03-17:
Seems like people are confusing this with a fan spin direction setting. Let's change the text to "Invert PWM Duty Cycle"

skot, 2025-03-17:
or "Invert Fan Duty Cycle"

JasonB1833, 2025-03-17:
Hey there, you just need the "invert fan polarity" calls to change for clarity throughout the codebase?? just asking for clarification before I jump into it and blindly change every instance of it to a clearer name.

skot, 2025-03-17:
> Hey there, you just need the "invert fan polarity" calls to change for clarity throughout the codebase?? just asking for clarification before I jump into it and blindly change every instance of it to a clearer name.

The only change needed is the text on the settings tab of AxeOS:

[image]

skot, 2025-03-17:
and lets go with "Invert Fan Duty Cycle"

JasonB1833, 2025-03-17:
> and lets go with "Invert Fan Duty Cycle"

good to go, pr is up

PR #779 (2025-03-20): fix: changed 'Invert Fan Polarity' text 
- changed above text in the settings menu to "Invert Fan Duty Cycle" for added clarity
closes #776
closed 2025-03-20 by WantClue