bitaxeorg/ESP-Miner #785: v2.6.0b11 bricks 403 board

jusincase99, 2025-03-21:
The v2.6.0b11 update causes the 403 board to fail to reboot at all. All there is is a black screen and no activity at all.

Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots & Photos**
If applicable, add AxeOS screenshots and/or photos of your Bitaxe to help explain your problem.

**Hardware (please complete the following information):**
 - Bitaxe HW version: [e.g. Ultra 205, Supra 401, etc]
 - Bitaxe HW vendor: [Where you purchased the Bitaxe, or self-built]
 - ESP-Miner FW version: [e.g. 2.1.1, etc]
 - Hash Frequency:
 - Voltage:
 - Pool URL, Port, User:

**Additional context**
Add any other context about the problem here.

jusincase99, 2025-03-21:
> Uh oh! That's not good. I'll take a look today. Can you tell us what method you used to flash v2.6.0b11?
>
> Is there any log output?

I used the bitaxetool method with a usb c data cable connection. I didn't collect an output log, not sure of the commands to use to get one. I could re flash if you can tell me the bitaxetool commands to use to generate the log?
~All the best

skot, 2025-03-21:
If you're already connected over USB you can just open a serial terminal at 115200 baud.

jusincase99, 2025-03-21:
> I'm not, I flashed back to the previous 10 build and am over wi fi so I have to rehook it up and open the bitaxe tool.

No error log when flashing, appears to finish flashing goes to the restart then c prompt just fine but then the 403 just has a blank screen, fan won't run even after pressing the restart and trying an unplug form the electric and replug it back in.

jusincase99, 2025-03-21:
Here are screen shots for you[Screenshot 2025-03-21 144342.png][Screenshot 2025-03-21 144225.png][20250321_144447.jpg]

jusincase99, 2025-03-21:
Screenshots of reflash back to ob10, 403 working perfectly with that build.[20250321_145823.jpg][403 board v10.png]

jusincase99, 2025-03-21:
0b12 seems to be working perfectly. Thanks!

closed 2025-03-22 by jusincase99