bitaxeorg/ESP-Miner #788: add TPS546 fault checking to selftest

skot, 2025-03-22:
I think it would be useful to check and report the status of all the TPS546 faults and warnings during the selftest.

skot, 2025-03-22:
this can be as simple as calling VCORE_check_fault() somewhere in selftest and printing the result to the log. Just gotta figure out where in the selftest is the most useful.

JasonB1833, 2025-03-22:
I threw VCORE_check_fault() just after the voltage regulator testing as I figured these two should be close together since they are related to voltage regulation anyways, not sure if this exactly what you're looking for but I'm open to suggestions, this just seemed like the quickest answer

[image]

skot, 2025-03-22:
> I threw VCORE_check_fault() just after the voltage regulator testing as I figured these two should be close together since they are related to voltage regulation anyways, not sure if this exactly what you're looking for but I'm open to suggestions, this just seemed like the quickest answer

I didn't know you were working on this.. I just made PR #789. Turns out that VCORE_check_fault() only fails if there is a I2C error with the TPS546. I had to also look at GLOBAL_STATE->SYSTEM_MODULE.power_fault for the test result.

JasonB1833, 2025-03-22:
@skot all good, I didn't announce my presence because I had fairly limited time and wasn't sure if I could get it done in a timely manner. regardless it gave me more time to delve into the codebase which can only help.

closed 2025-03-24 by skot