bitaxeorg/ESP-Miner #811: BUG:  Where did the average hashrate line go?

jtsmith0101, 2025-04-01:
**Describe the bug**
Since at least version 2.5.0, the histogram for the asic hashrate included an average hashrate line.  This appears to be missing in 2.6.1.

**To Reproduce**
Update to 2.6.1

**Expected behavior**
Unless discontinued, I expected to see the hashrate, asic temp, and an average hashrate (I guess this was computed and displayed a rolling 30 minutes or so)

**Screenshots & Photos**
If applicable, add AxeOS screenshots and/or photos of your Bitaxe to help explain your problem.

**Hardware (please complete the following information):**
 - Bitaxe HW version: [Gamma 601]
 - Bitaxe HW vendor: [Where you purchased the Bitaxe, or self-built]
 - ESP-Miner FW version: [2.6.1]

[image]

mutatrum, 2025-04-01:
It moved to the card at the top:

[image]

See #633 for context.

jtsmith0101, 2025-04-01:
Ha!  Fair enough

closed 2025-04-01 by jtsmith0101