bitaxeorg/ESP-Miner #817: A small idea

Uefi1, 2025-04-02:
Hello, I think it would be great to have the option in the settings to split the hashrate of an ASIC across different pools and coins. For example, if I have a speed of 500 GH/s, I would like to be able to allocate 100 GH to 5 different coins, such as BCH, eCash, Space, Fractal Bitcoin, and Peercoin. Or, for instance, allocate 250 GH to two coins, like BCH and eCash.

This would increase profitability because when mining in PROP or PPLNS mode, the reward per block would still be just 1 cent, regardless of whether the device submits shares at 500 GH/s or 10 GH/s. The smallest share will always determine the payout, so having the ability to split hashrate efficiently would be beneficial.

skot, 2025-04-14:
This seems like a lot of complexity for minimal benefit -- especially on single chip, low hashrate miners like Bitaxe.

MyOwn2C, 2025-04-14:
This doesn’t make sense. 
Bitaxe is already “slow”
Splitting hash rate will make it even slower. 
A lot of pools will reject slow hash rate devices. 
This will eliminate Bitaxe to participate in lots of pools. 
If a pool accepts slow devices, it is not a pool you want to be in anyway.

Uefi1, 2025-04-14:
> This seems like a lot of complexity for minimal benefit -- especially on single chip, low hashrate miners like Bitaxe.

Honestly, I don't see any difference between 500 GH/s and 100 GH/s — in either case, the coin's required difficulty will never be reached, and it's better to rely only on the guaranteed block every 10 minutes !

zcht, 2025-04-14:
Simple solution: buy two (or more) Bitaxe for pool A or B.

Uefi1, 2025-04-14:
> Simple solution: buy two (or more) Bitaxe for pool A or B.

Hi, well first of all, it costs a significant amount of money — enough to buy a pretty good ASIC with 100–200 terahashes. And secondly, even if we're just talking about one BitAxe, it costs as much as two or three Antminer S9 units, each with 14 terahashes, which is honestly not reasonable at all.

closed 2025-04-14 by skot