bitaxeorg/ESP-Miner #829: Password for the web interface

Uefi1, 2025-04-08:
Hello, this morning I woke up and discovered that almost all my settings were changed. Someone accessed the web interface somehow. Please add authentication to the ASIC's web interface.

MyOwn2C, 2025-04-08:
You should secure your wifi network first.

Uefi1, 2025-04-08:
> You should secure your wifi network first.

My Wi-Fi network is secured, and no external access to other interfaces is possible.

closed 2025-04-08 by WantClue

mdklapwijk, 2025-06-18:
It would be an additional barrier/layer of security, for those situations where you find yourself with a compromised client in your network; e.g. a worm/rootkit/malware which automatically searches for esp-miners to change the address/stratum-user. Yes, if your network is compromised you have bigger issues, but password protection combined with a max attempt count would create some extra hurdles.

By the way, this reminds me an awful lot about the discussions about the storage of unencrypted, as in plain text, passwords in [Filezilla](https://www.bleepingcomputer.com/news/software/filezilla-ftp-client-adds-support-for-master-password-that-encrypts-your-logins/)...