bitaxeorg/ESP-Miner #833: Building binary files for firmware yourself.

vaavdeev, 2025-04-10:
Building binary files for firmware yourself.

I have:
- Bitax Gamma 601
- esp32s3
- Windows 11
- esp-idf-v5.4.1
- Visual Studio Code 1.99.1
- Python 3.11.2
- Noda v22.14.0.
- Ninja version ("1.12.1")
- PowerShell 7.5.0

I cloned the project
https://github.com/bitaxeorg/ESP-Miner

Made a Build Project from the original files.
I didn't change anything.

Received my files:
- www.bin
- esp-miner.bin

www.bin - everything is fine. Its size corresponds to the original developer.
esp-miner.bin - **1 149 KB**. Original file size - **1,163 KB**

There are problems with applying Share deviations on the Dashboard screen.
Problems with displaying Power and input voltage.

The device is mining.

Can you tell me what could be the problem with the build?
Thanks.

skot, 2025-04-10:
Did you use bitaxetool to write a cvs file to the ESP32 nvm? Can you post some screenshots of the issue?

vaavdeev, 2025-04-11:
> Did you use bitaxetool to write a cvs file to the ESP32 nvm? Can you post some screenshots of the issue?

I don't quite understand what you're talking about.
I receive two binary files in my program.
I flash the device in the standard way through the browser.
I upload my files.

I am sending a photo of the problem with displaying parameters.

[image]

[image]

mutatrum, 2025-04-11:
Can you clear the browser cache and reload the page? Might be some old stuff lingering the in browser?

vaavdeev, 2025-04-11:
> Can you clear the browser cache and reload the page? Might be some old stuff lingering the in browser? Just want to rule out that. I do all my updates through self-built firmware, so they should be ok.

Refreshing the page didn't help.
The problem, as I wrote above, is that I am building a binary file different from the size of the original.
I can't help you figure out the reason.

skot, 2025-04-11:
> I don't quite understand what you're talking about.
> I receive two binary files in my program.
> I flash the device in the standard way through the browser.
> I upload my files.
>
> I am sending a photo of [...]

Ah, ok I thought you were flashing a new device.

What version did you have loaded before flashing your own self-built images?

Are you building from a locally cloned git repo? Your version number looks strange.

STSMiner1, 2025-04-12:
@vaavdeev 

VS Code, bottom of the window, do you see this ?

[image]

WantClue, 2025-04-16:
please move this over to the discussion tab, introduced it today to this repo 🚀

closed 2025-04-16 by WantClue