bitaxeorg/ESP-Miner #841: Building FW bin: boot_comm: mismatch chip ID, expected 9, found 0

dskvr, 2025-04-12:
I have been building and deploying `ESPMiner` for a couple months now. However, recently, when attempting to load the successfully built binary [`esp-miner.bin`] I get the error `boot_comm: mismatch chip ID, expected 9, found 0`

I was making some changes to the firmware, and so assumed it was my fault, but when checking out `tags/v0.6.3` and building it, I am getting the same error. If download the pre-built binary, presumably built from `tagsg/v0.6.3` attempt to load it, the firmware updates without this error; so unlikely this is a device/gui issue.

I cannot find the error string "mismatch chip ID" anywhere in the source code, so I am assuming this is an `idf` error. 

I am pretty sure I have read all documentation on building from source, and have attempted to find anyone having the same issue to no prevail which is why I am posting this issue.

Possibly related to #833

mutatrum, 2025-04-15:
The log line is here:

https://github.com/bitaxeorg/ESP-Miner/blob/ec57370659cfb3a711ef70424aff74ac755c5054/components/asic/common.c#L68

Can you make sure you're building for the ESP32-S3?

WantClue, 2025-04-16:
please move this over to the discussion tab introduced today 🚀

closed 2025-04-16 by WantClue