bitaxeorg/ESP-Miner #908: Actual Bitaxe Consumption is Underestimated

cicoph, 2025-05-09:
I ran some tests on Gamma using a smart plug to measure actual power consumption. From my tests with the default settings, I got the following results:

Gamma: 14 W
Plug: 24 W

With a bit of overclocking:
Gamma: 21 W
Plug: 35 W

Is it possible that there’s a bug in the power measurement on Axeos?

I repeated the test with different settings and other systems to check for measurement errors of the smart plug. I found that the plug with the power supply turned on consumes 3 W. However, when doing the calculations, the actual power output shown on Axeos still doesn’t add up. Has anyone else run similar tests?

Nexus9090, 2025-05-09:
I'm running a Gamma also fractionally overclocked 550Mhz.

The results at the plug you're seeing are due to the inefficieny in the mains power supply, this varies significantly with the quality of the power supply brick and if it has synchronous rectification or not. The low/mid end 30W supplies tend to only be ~80-85% efficient with some of the better ones achiving 85-90%

With the PSU I have I'm seeing  
Gamma: 17W  
Plug: 20W
Off load the PSU draws less than 0.4W

With the results you mention, it suggest the PSU you have is low end and low efficiency.

If every last watt matters to you, you might want to look at getting a GAN type power supply as these offer the highest efficiency typically in excess of 90% efficient. Most of these seem to be offered with USB-C connectors, so an adaptor will be needed to provide power to the BitAxe.

Its worth shopping around a bit and checking the specifications in detail, they can be a bit missleading.

NilByte, 2025-05-09:
I'm running underclocked 494MHz, asic voltage 1.02V on a 30W PSU.

Gamma: 15W
Plug: 17.1W

Check if you run the latest firmware. There was a release increasing the displayed power consumption by 5W.

mutatrum, 2025-05-10:
Silly idea maybe, but this could to be user configurable as well. Current devices can only guess what the extra consumption is, and if it's configurable one can check with a wall meter.

seepv, 2025-05-10:
Also an issue, for the Plug, could be the kind of measurement.

Small wattage and e.g. inductive loads, are problematic for some plugs.
The Tasmota-Plug of mine measures 0W with an v204 only attached.