bitaxeorg/ESP-Miner #911: Stop mining if pools are not reachable

mutatrum, 2025-05-12:
We had an internet outage, and the result was the following:

[image]

In this case it would be prudent to stop mining, as it doesn't make sense to keep hashing the same work over and over again.

skot, 2025-05-12:
I'd love to get this one fixed soon. I made a "status task" issue a while back. 

My thought is that we have high priority task, running on a long interval that check to make sure mining is happening; stratum server is still sending work, ASIC is still sending shares, vreg is still enabled, etc.

If anything is amiss we should, at the very least show the alert in AxeOS and Bitaxe display.

mutatrum, 2025-05-12:
#272

RealHumanWithHair, 2025-05-12:
> We had an internet outage, and the result was the following:
> 
> [image]
> 
> In this case it would be prudent to stop mining, as it doesn't make sense to keep hashing the same work over and over again.

From where did you buy your bitaxe?

RealHumanWithHair, 2025-05-13:
Should be nice if there have some kind a method with time limit per task, something like 1-2min even 5min to be safe.