bitaxeorg/ESP-Miner #912: Identify miner from AxeOS

mutatrum, 2025-05-12:
For people with lots of Bitaxes, it would be nice to be able to trigger a big visible marker on the display from AxeOS and the swarm page. Similar to `Identify monitor` in Windows or what some smart thermostats can do:

[image]

Basically just flash something on screen for 5 or 10 seconds or so.

STSMiner1, 2025-05-12:
5 or 10 seconds might not be enough, all depends on where they are located (takes 30 seconds to walk to the first device and then another 30 seconds plus to check all the others).

skot, 2025-05-12:
We could have the screen invert until the boot button is pushed?

Then we just have to make sure people who inadvertently got in this mode know how to get out...

STSMiner1, 2025-05-12:
Could set it as invert for 30 seconds and then it switches back automatically. 

If using a message that gets displayed .... like ...

"Hey !  I'm the device your looking for !"

NilByte, 2025-05-12:
https://github.com/bitaxeorg/ESP-Miner/issues/615