bitaxeorg/ESP-Miner #939: Custom nonce and extra nonce range

precious8821, 2025-05-20:
Is it possible to modify the firmware or software so that a custom nonce range and extra nonce value can be used? So that way it could run a nonce range starting from say 2 billion to 4 billion, increasing by 1 per hash, and once 4 billion is reach then starting again at 2 billion, cycling through. All WHILE having a custom EXTRA NONCE range starting at say 17,446,744,000,000,000,000, or whatever custom value is entered not exceeding 2^64. And then every time all the nonces in the nonce range are all hashed and restart at the preset custom value of 2 billion, then the custom extra nonce value will increase by one and try all the nonces in the defined nonce range again. Repeating this process again and again.

If this modification is possible, could it also be possible to enter these custom range values within the web portal for the miner, rather than having to hard code the custom values in the firmware itself.   So that way the values won't have to be entered in the software code itself every time, and instead the custom nonce range and extra nonce range values can be entered into the admin screen through the web portal of the miner.

mutatrum, 2025-05-21:
Sounds like some overlap with #824.

What's the rationale of wanting to choose the values?

g1ass1, 2025-05-21:
I have the same ask. - but more so on the nonce... - I had implemented MQTT pub and sub logic to achieve changing the extra nonce range. but I am struggling in the firmware to find where the nonce range is controlled.  there are a number of other use cases here where mqtt could prove usefull.

I'm not so much changing the sequential logic, (although control here would be nice) my ask is more basic on nonce - a start and end nonce to limit the logic to certain ranges.

I have been able to implement a range of various strategies for extra nonce control but struggling to find where to control the nonce is controlled.

if you could point this out in the code base, I'm happy to make my own local amendments and go from there.

precious8821, 2025-05-22:
G1ass1, could you share the coding you used to adjust the extra nonce range? Does your modification allow for a custom extra nonce starting value to be used, of course not exceeding 2^64?   Also, please share the file names where you added or modified the coding for the extra nonce.  Thank you.   Hopefully someone might have an answer to adjust the nonce range as well.

g1ass1, 2025-05-22:
uint32_t min = 0x33333333; // Top 80% of 32-bit range
        uint32_t max = 0xFFFFFFFF;
        while (GLOBAL_STATE->stratum_queue.count < 1 && GLOBAL_STATE->abandon_work == 0)
        {
            if (should_generate_more_work(GLOBAL_STATE))
            {
                // Generate extranonce_2 using double SHA-256 (Bitcoin style) of job_id, merkle_root, timestamp, and rand()
                char key_material[160];
                snprintf(key_material, sizeof(key_material), "%s_%s_%llu_%lu",
                    mining_notification->job_id,
                    merkle_root, // Use the previously calculated merkle_root for this job
                    (unsigned long long)esp_timer_get_time(),
                    (unsigned long)rand());
                uint8_t hash1[32], hash2[32];
                mbedtls_sha256((const unsigned char*)key_material, strlen(key_material), hash1, 0);
                mbedtls_sha256(hash1, 32, hash2, 0);
                // Use the last 4 bytes of the double SHA-256 for extranonce_2
                uint32_t extranonce_2 = (*(uint32_t*)&hash2[28]) % (max - min + 1) + min; // Ensure in top 80%
                generate_work(GLOBAL_STATE, mining_notification, extranonce_2);
            }
            else
            {
                vTaskDelay(100 / portTICK_PERIOD_MS);
            }
        }

these were the lines i amended in the create_jobs_tasks.c
I am working on a "bee-hive" approach to swarm an area.  this function selects the extranonce2 as a double to randomise selection of extranonce - the goal is to set a center, and a upper and lower window to limit search to increase focus on an area and randomly swarm it with activity, of nothing found back off, hit found center and focus more

this version just selects random exranonce hashes in the upper 80% of the range

g1ass1, 2025-05-22:
> Sounds like some overlap with [#824](https://github.com/bitaxeorg/ESP-Miner/issues/824).
> 
> What's the rationale of wanting to choose the values?

bump on this?

precious8821, 2025-05-23:
G1ass1:  on the extra nonce in your code you provided, it looks like you are only using 32 bits, with 4,294,967,295 as the max value for the extra nonce , i.e 2^32.   However, since the extra nonce can be up to 8 bytes, or 2^64.  Shouldn't your extra nonce max value should be closer to 18,446,744,073,709,551,616 ??   Only the nonce value is limited to 2^32, not the 'extra nonce'.   So the 80% top of extra nonce range should have a minimum value of 14,757,395,258,967,641,292 with a maximum value of 18,446,744,073,709,551,616.   Let me know your thoughts.

Also, please share your 'bee-hive' approach coding for the extra nonce when you complete it, as that does sound interesting to try out.

Nexus9090, 2025-05-23:
It seems to me that it would be useful to implement a few options in terms of the extranonce range and search patterns.

1. Linear,  extranonce starts at zero incrementing by one each time (I'm not sure but I think this is how it works presently, which means a large amount of the extranonce space is never explored)
2. Inverse Linear,  start at 2^64 and decrement by one each time
3. Swapped linear, a combination of Linear and Inverse linear on each alternate extranonce (one at the bottom, next one at the top alternately)
4. Swapped linear within range, same as 3 but specifying the start and end values of extranonce
5. Random within range, so specify an extranonce range that you'd like to search with a random search pattern
6. Max random, random extranonce over the whole extranonce 2^64 range 
7. Random base with linear, start point is random in the 2^64 range but search is linear from that point

precious8821, 2025-05-26:
G1ass1:  on my last comment how I explained how I plan to set the extra nonce by taking the take the global network's hash rate in hashes/second times 60 times X gives around the total hashes tried to find a block in X minutes. Then set the starting extra nonce value to that amount.   I forgot to say to also divide that number by the total nonces of 4,294,967,296., and then set that calculated value as the extra nonce.   So that will estimate theoretically the hashes performed in those 10 minutes, if you set X to 10.   To summarize, the plan is to take the (global network's hashes/s times 60 times X)/4,294,967,296, with X being the minutes it takes to mine a block, so set X to maybe 10 or 5 or whatever the average is for the last most recently blocks mined.  And then set that value as the starting extra nonce.  

I look forward to seeing your beehive attack script when its ready.   

 Also, let me know if you know if bitaxe can mine directly to bitcoin core's bitcoind.   That way there won't be the issue of the extranonce being partially preset by the mining pool, since the value will be sent straight to bitcoind.

Nexus9090, 2025-05-26:
> G1ass1, thanks for the info. Do you know if bitaxeorg can mine directly to bitcoin core's bitcoind, that way the extra nonce value won't be partially limited by the pool server? My reason for wanting to modify the extra nonce is as follows. To take the global network's hash rate in hashes/second times 60 times X gives around the total hashes tried to find a block in X minutes. Then set the starting extra nonce value to that amount.

Presently ESP-Miner only mines to Stratum pools, I did ask if there was any development plans to allow direct to bitcoin core mining here :- #901 

@mutatrum seems to have a better grip on the technical details, so hopefully they'll come up with a workable solution for mining direct to bitcoin core.

mutatrum, 2025-05-26:
I'm not 100% sure exactly which parts are provided by the pool, which part ESP-Miner handles, and which part is handled inside the ASIC. It's also slightly different between ASICs.

There is some related discussion on this here: https://discord.com/channels/1091348375301013615/1094385611718270977/1364651145095548959

precious8821, 2025-05-26:
I accidentally clicked the close issue button again.   Just reopened it.

g1ass1, 2025-05-27:
Extranonce2 isn't the issue, it was fairly easy to discover how to do and implement, I have a random generation up and stable now for about a week (see above) - the sky is the limit here tbh - I used sha hash and Merkle root and job id changing and timestamps as seeds to get a truly random number as esp_random has issues here. - plus defined the upper and lower limit in the range.  the code isn't that clean, but it does work. 

This has enabled me to work on a build at the moment that defines a center (focal point) and dynamically increase or limit search renomination on areas that hit with a certain difficulty target / increase in difficulty. - MQTT PUB/SUB will come later to visualize this space in Grafana and dynamically change these values from the outside - will follow up when I have that finished, but likely to be mid June/July. (beehive logic I discussed earlier).  I have an mqtt broker coded on the board at the moment that sends data to my home assistant mqtt broker.

the idea overall is that while SHA has proven to be completely random there is VERY SIGNIFCANT structure built into how we all approach this problem - i.e start at the bottom and increment up, and that pools between themselves don't co-ordinate in any way between pools - so we are all likely re-hashing the same space over and over again leading to duplicated and this wasted hash power. - particularly in the first 3-5 mins of job start ( while i recognize pools will organize ad split up work to a point - we all start at the same spot, with broadly the same timestamp, with broadly the same logic - so it stands to reason there is ALOT of overlap because POOLS don't organize between themselves).  Think of a perfect TV image slowly degrading into white noise over time, white noise being ideal state - so why not start there.  This kind of implantation, while it statistically won't increase chances, practically it will as I'm not duplicating over global workload and providing truly unique hash results to the network/.

My issue to complete this was controll over the nonce, limiting the range to a predefined start and stop. eg, saying, start at 0, but stop 25% of the way through the range, eg 0x33333.... and or start at the top, and increment down, and combine that with the extra nonce randomization.  right now start id defined, and stop is defined as bit overflow, then it increments extranonce 2 per logic and repeats.

the Protocall suggests a nonce start, and stop location is possible, but I'm having difficulty getting it to work.  bm_jobs is the function where it would be but having trouble implementing the stop digit.  bit when i did, the board stops hashing.  suggesting it either doesn't work, or mor likely I did it wrong.

precious8821, 2025-05-31:
Until the bitaxe is made to directly mine to Bitcoind's RPC, it does seem that setting a custom extra nonce range or even randomization, won't make too sense or could be a problem, due to the mining pools partially setting their own extranonce values in the coinbase.  Here is part of the problem.  The pools even use part of their own extranonces as  'coin base 1 & 2' values nearly as an extranonce signature unique to their mining pools.  Technically, according to Bitcoind's RPC guidelines, the extranonce value can be up to 100 bytes, not only 8 bytes. But with the pools, they are setting their own random partial extranonces, so it's hard to say how much of the 100 bytes available they are using up for their own partial extranonce values,  which includes their coinbase 1 & 2 values,  all of which is placed for the coinbase transaction.   Therefore, if you randomize the extranonce or set it on a custom range with a high value, you could end up exceeding the 100 byte maximum for the coinbase transaction if the mining pool is using a lot of bytes for their partial extranonce values.  Unless you set some type of limiter so that your custom extranonce plus the pools own partial extra nonce doesn't exceed 100 bytes.

Nexus9090, 2025-05-31:
Well, my board has been running with random extranonce_2 for the past 4 days or so with a couple of interruptions while I was testing software developments and I can report that it's made little or no noticable difference to the returned share values. 

The best share during that time was 110M with most shares in the 1-10K region and occasional in the 10-500K region. So, realistically nothing different from normal.

Looking at all of the technical challenges of controlling extranonce in a meaningful way the overheads of developing something probably outweighs the benefits.

It would be nice to see ESP-Miner have a GBT function and full control over extranonce and hopefully someone will implement it. However that's not going to be me.

**_Hypothetically and unproven_** a thing that does bother me is that when a share of low value is submitted, rather than carry on from where it was found it appears to start a new job. So within a given hashing range its possible you'll get a share above pool difficulty right away not knowing that the very next hash in line is the one that wins it by which time your miner has been instructed by the pool or pre-prepared work to go look elsewhere. 

Its a bit like counting to three then being told to start again from 4billion. there's a big gap inbetween. Its a bit confusing really.

Anyway, that's me done on this topic for a while.

Good luck working it all out.

adammwest, 2025-06-02:
> id love to understand if limiting of the nonce range is possible.

read PR420 in detail it contains information about size and timing for nonce ranges
https://github.com/bitaxeorg/ESP-Miner/pull/420] 

you need a gamma to use start location

use the  starting nonce from a  job paramater, for the bm1370 it works or you can use CNO register for start aswell
```c
int start_nonce = 512;// a number between 0-512
int start_nonce_n0 = (uint32_t)(start_nonce & 0xff;
int start_nonce_n1 = (uint32_t)(start_nonce) & 0xff;
next_bm_job->starting_nonce =  (start_nonce_n1 << 8) + start_nonce_n0;
```

for the size use hcn register 0x10 PR420 has the details and functions
```python
size = 512 # number between 0 and 512
percent = size/512
hcn_val = percent *256*256*256 * 25/(freq_mhz) 
timeout_ms = percent* 2**25 /(freq_mhz*1000)
```

nonces_rolled = size*2**16 (each core rolls this)
nonces_in_parallel = 128 (each core gets 1)

that will control the bit range
[0-7] core id
[7-16] controllable 
[16-32] rolled per core

if you use other BM chips (BM1362/66/68) only the HCN is usable meaning you get a 
0->size control only

johnnyBytes66, 2026-09-23:
I know this is an old thread, but there is something that you may have not realized in your investigation into the HCN register (0x10). I see the 0x10 register as a timer on the Version Rolling feature only. And this directly affects how far into the nonce space that hashing will go before the next Version is processed according to the ASIC's set frequency of course. And so on, until the hashing time limit has been reached and a new job is started. Increasing 0x10 register value will hash deeper into the nonce space. And lowering the 0x10 register value will hash less of the available nonce space before rolling the version. Increasing the hashing frequency will also hash deeper into the available nonce space without needing to change the 0x10 register.

However, both the BM1366 and BM1368 are nonce limited in their hashing nonce space. And the simple formula is like this:

For BM1366: (core_count / 128) * 100 = (112 / 128) * 100 = 87.5% (12.5% not available)
The highest possible nonce for the BM1366 is 0xDFFFFFFF

For BM1368: (core_count / 128) * 100 = (80 / 128) * 100 = 62.5% (37.5% not available)
The highest possible nonce for the BM1368 is 0x9FFFFFFF

The BM1370 and newer ASICs don't have this limitation as far as I have tested.

johnnyBytes66, 2026-10-13:
**How to calculate Register 0x10 value:**
To calculate correctly for the Register 0x10 value, is a 2 part process.

**Step 1:** First determine the nonce depth per Version Rolled you want the miner to cover. This is a percentage (%) of the 2^32 full nonce space. Then divide this number into the hash_rate * job_time_seconds of the miner. The result is the number of Versions that will be rolled for this job_time and is needed in step 2 to calculate the actual Register 0x10 value.

**Step 2:** The simple formula to calculate the Register 0x10 value is:
(ASIC_job_time_in_ms / ASIC_VR_const) * 25,000,000 / versions_rolled = reg_0x10;

**Example:** Lucky Miner LV08 Pro @575 MHz (BM1368 * 9)

**STEP 1:**
nonce_depth = 60.06% * 2^32 = 2,579,557,357.9776
BM1368_small_cores = 1,276
ASIC_count = 9
ASIC_job_time_seconds = 1.0
total_job_hashes = 575,000,000 * 1,276 * 9 * 1.0 = 6,603,300,000,000
versions_rolled = 6,603,300,000,000 / 2,579,557,357.9776 = 2,559.86

We round up versions_rolled to 2560 (0xA00)

**STEP 2:**
ASIC_VR_const = 125 (125 for BM1368 and BM1370, 250 for BM1366)
(1000.0ms / 125) * 25,000,000 / 2560 = 78,125 (0x1312D)

So for a nonce_depth of 60.06% (0.6006) and a job_time of 1.0 second (1000.0ms) for the above example, we calculate 0x1312D as our reg_0x10 value.

**NOTE:** The BM1368 has maximum nonce_depth of 62.5% (see previous post). Hashing past this maximum will reduce the expected hashrate. On a single ASIC miner, no duplicates will be submitted unless all Versions are rolled and a wrap-around occurs. On multi ASIC miners, duplicates will occur if either maximum allowed nonce_depth is passed or Version wrap-around occurs.

**ALSO NOTE:** That ASIC Version Rolling has a granularity of 16. This means that lowering or increasing the reg_0x10 value slightly will not make a difference in the amount of Versions Rolled.

Keep this in mind when calculating Register 0x10 value:
BM1366 max nonce_depth = 87.5% of 2^32
BM1368 max nonce_depth = 62.5% of 2^32
BM1370 max nonce_depth = 100.0% of 2^32 (no limitation)

If the miner's hashing frequency is changed, the reg_0x10 value should be recalculated as increasing the frequency can push hashing past the max allowable nonce space. Changing the duration of the ASIC_job_time does not affect the nonce_depth, it simply increases or decreases the amount of Versions Rolled for the job_time. Although too long a job_time and Version Rolling will wrap-around and hence cause duplicates.

However on Multi-ASIC miners, reduce nonce_depth to about 1% to 3% from max allowed as there is a slight delay in propagating the data down the ASIC chain. Not doing so will cause occasional duplicates. Single ASIC miners don't have this issue.

adammwest, 2026-10-14:
your version
```
hcn_register = (1000/125) * 25000000 / (ASIC_freq_hz * ASIC_small_cores * ASIC_count * ASIC_job_time_seconds/ (nonce_percent * 2^32 ))
hcn_register = 78,125
```

my version
```
hcn_max = 2^32 / 128 / 16 * 25/575 * 0.5
hcn_max = 45590 (for 16 asics, 100% nonce range)
max_timeout = 65536/(2048/128) * 2^32/128/16 / (575000) = 14.9s (16 chips, 100% nonce range)

hcn_register = 0.6 * 45590 = 27354
timeout = 0.6* max_timeout = 8.96s
```

```
hcn = (nonce_percent *  2^32)  * (25 / freq) * (0.5 / asic_cores_up / asic_count_up) 
hcn = (nonce_percent * 2^32)  * (25 / freq ) *  (1000/vr_const)  / ( small_cores * count * job_time)
```

the disagreement is how you assume division of the space you assume perfect division, I assume power of 2 reservations
```
 0.5 / 128 / 16       =  0.00024414062 (mine, assuming 16 chain length reserved)
 0.5 / 128 / 9         =   0.00043402777 (mine, assuming 9 chain length reserved)
 8 / ( 1276 * 9 * 1) = 0.000676 (yours)
```

if you can execute  78,125 without duplicate hashrate it means  45590 * 16/9 = 81048 is max
it means lucky miner properly divides the asic chain space for older chips
if this is the case (it would be a new information) its not so relevant because lucky miner is closed source so we cannot add something yet. However it creates a new direction for this type of research, in Bm1370 the CNO register can divide by non power of 2 offsets.

johnnyBytes66, 2026-10-14:
@adammwest 

I am using the same formula to calculate the 0x10 value on Ultra, Supra, Gamma, GammaTurbo, SupraHex, LV07, LV08 and LV08 Pro. I am also using an older version of AxeOS from the Tiny Chip Hub which I have heavily modified. All my clones including LV07 (BM1366 * 2), LV08 (BM1366 * 9) and LV08 Pro (BM1368  * 9) have modded AxeOS installed on them (I am not using any locked down firmware). I have added code that allows me to change on-the-fly: ASIC_job_time, Version_rolling_start, Version_rolling_quantity and nonce_depth. I can monitor version rolls covered in a job_time and on every combination I have tried, I get 100% accurate results. With fine tuning I can eliminate all duplicates from any miner whether it's a single ASIC or multiple.

In the last 6 months to a year, I have tested several hundred different 0x10 settings on multiple miners so I have a lot of trial and error test data that has enabled me to work out an accurate formula that seems to work flawlessly.

I checked on the LV08 Pro at 62.49% nonce_depth and my 0x10 value is 81,292 (you calculated 81,048). Only a 0.3% of a difference. Using my formula for exactly 62.50% is a little more at 81,301.