bitaxeorg/ESP-Miner #993: Unable to reach Minimize Logs button once the log text fills the screen

xr1140, 2025-05-31:
The ‘Minimize Logs’ button should remain visible on the screen at all times. Currently, when the screen fills with log text, the button disappears from the visible area of the browser and is difficult to access. It would be helpful for the button to stay on the screen permanently, and for the ESC key to function as an equivalent to clicking it.

The ‘Stop Scrolling’ button should also remain visible on the screen at all times.

duckAxe, 2025-06-01:
I tested the close button several times bevor PR. Now I tested the button again and it works as expected. Have you guys tried it on desktop or mobile?

https://github.com/user-attachments/assets/d928dfbd-eae8-4f09-b1e1-0356e11a68fd

duckAxe, 2025-06-01:
The only difference between me and you: Windows. Unfortunately, I don't have Windows to test ;(

STSMiner1, 2025-06-01:
I tested this on my android phone, when in fullscreen mode the icon does not "float" in the top right corner, it stays at the top when there is a lot of data presented.

[image]

STSMiner1, 2025-06-01:
Set as an 'overlay' in this example (mockup) so that stay visible on the screen.

[image]

duckAxe, 2025-06-02:
I can't fix things until I can reproduce them on my local environment. I have just tested on Firefox and Chrome, works. 😢

duckAxe, 2025-06-02:
Thank you guys for help. I definitely need a real Bitaxe, not just a local environment ;)
The PR #1000 ✅ is your friend.

skot, 2025-06-02:
> Thank you guys for help. I definitely need a real Bitaxe, not just a local environment ;)
> The PR #1000 ✅ is your friend.

I have some Bitaxe available for developers; hit me up on OSMU Discord!