bitaxeorg/ESP-Miner #998: Feature Suggestion

carliartger, 2025-06-02:
Please integrate the following Features:

Generell:
- Remote Access / Control to the webinterface or App / Bitaxe Gamma 601 
-  With allert function for overheating 

Webinterface / 
- Option for shut down / deactivate 

Trank you

carliartger, 2025-06-03:
Right I Understand! Great Suggestion with Same WLAN SSID and PW. It works.

Regarding the alert:
I just wanted some Info and Control if i am on travelling and the Heat is over 80 (depents on the weather - Room teperature ) , so that I can Change the Settings Parameter of the Bitaxe from Remote. 

My goal is to run the axe 24/7 . 365 days with Max uptime and best conditions. But there are some possobilitys that can influce this.

I will Check wich information a secondary service could get from the axe .

Thank you all Great Support .

WantClue, 2025-06-27:
This Issue will be closed, if more discussion is needed feel free to open a [discussion](https://github.com/bitaxeorg/ESP-Miner/discussions)

closed 2025-06-27 by WantClue