bitaxeorg/ESP-Miner #999: Feature: Asic work should be made from the newest stratum job

adammwest, 2025-06-02:
**Relevent quotes** 
this was discussed in PR420
skot 
```
It sounds like even if clean_jobs = false the pool does expect the miner to switch over to the new work (new JobID) in some amount of time.

Can we realistically switch on every mining.notify?
``` 

ben 
```
The miner should switch to new work sent by the pool as fast as reasonably possible to avoid missing out on new/high fee txs
```

**Problem**
Currently the stratum messages are put in a queue, then jobs are made by rolling extranonce2
a few jobs can come in, but work is still being generated from an old stratum job. because a single chip is quite slow the extranonce2 range will never be exhausted therefore the job wont change until clean_jobs=true.

**Solution**
Clear the work queue immidiatly after a new stratum job message comes in regardless of clean_jobs=
then only generate work from this new stratum message

**Benefit**
Higher transaction fees/newer tx in a block

**Status**
Awaiting others opinions before this is decided as a good approach

mutatrum, 2025-06-04:
Proof of concept code: https://github.com/mutatrum/ESP-Miner/tree/no-asic-task

I don't think this will work on chips without version rolling, as the total work time is too short.

0xf0xx0, 2025-06-05:
runnin the poc for a day on my gamma with various clock settings, pretty good so far [image]

mutatrum, 2025-06-05:
It should be tested on a Max, without version rolling. Don't think that'll work though.

closed 2026-01-21 by WantClue