bitaxeorg/ESP-Miner #1002: Unified firmware update

mutatrum, 2025-06-04:
Currently having a 2-stage firmware update is error prone and confusing. Secondary goal is to step away from the strict filenames. There are a few options to do this:

1. Merge `esp-miner.bin` and `www.bin` into a single binary file. This is a custom format, and needs some sort of simple header + size. One thing we could add into this header is a device marker, so we can do a validity check on that instead of the filename to prevent people of flashing the wrong firmware;

2. Have a zip file with both files. Same idea here, and it's trivial to add simple manifest file for device validity check;

3. Geet rid of `www.bin` and have the AxeOS files embedded into `esp-miner.bin`. I haven't looked at this in detail, but it's worth exploring this as well, as this makes the upgrade path way simpler. We can just forget about the spiffs partition when this is loaded onto the device.

The first two need a clear upgrade/downgrade path, and we need to supply both variants for a while, possibly adding to the confusion.

STSMiner1, 2025-06-04:
#910 should cover it for now with ref to people not updating with both files that was included in v2.8.0.

mutatrum, 2025-06-11:
Previous issue with similar ideas and some technical exploration: #308

duckaxe, 2025-08-08:
Related https://github.com/bitaxeorg/ESP-Miner/discussions/1179

WantClue, 2025-11-29:
Community agreement atm is to keep it in for potential alternative WebUI like AxeWellUi