bitaxeorg/ESP-Miner #1011: Show block header information in Axe-OS

mutatrum, 2025-06-06:
Seeing https://x.com/boerst/status/1930754454451847375, it would make a lot of sense to show block header information on Axe-OS somewhere. Maybe a full clone of stratum.work might be overkill, but at least show the miner tag (not sure what it's called) and possibly the merkle tree buildup would be interesting and educational.

mutatrum, 2025-07-03:
Some more inspiration: https://hodl.camp/genesis_block

skot, 2025-07-06:
Ooo I really like this idea! Showing live info about the candidate block being mined would be awesome for learning about mining and something that no other miner does.

bboerst, 2025-07-14:
If you decode the coinbase tx from these `mining.notify` messages, you can get a bunch of interesting data, including the `height`. [BIP34](https://en.bitcoin.it/wiki/BIP_0034) requires that the `height` be included within the coinbase tx; so that's in there if you want to dig for it.

To decode the raw coinbase tx from the `mining.notify` message, it's:
```
coinbase1 + extranonce1 + "00".repeat(extranonce2_length) + coinbase2
```

...so in the case of the `mining.notify` example in the description above:

- `coinbase1` = `01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff370392ca0d0004b27c666804e66ff2190c`
- `coinbase2` = `0a636b706f6f6c1365752f736f6c6f2e636b706f6f6c2e6f72672fffffffff037a7c93120000000016001480bded37e3f86a1a546e099b15e2b02855d3843cfd0c61000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed2aaf372eab3aa252e8b72245d385e68a932bf06cf5c06ed51d013f6599d62d5b00000000`
- `extranonce1` (the pool gives you this during `mining.subscribe`, but only the length matters) = `00000000`
- `extranonce2_length` (the pool gives you this during `mining.subscribe`) = `8` (this is the byte length, so `8` bytes is `0000000000000000`)

Concat these ^ and you get the raw coinbase tx:
```
01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff370392ca0d0004b27c666804e66ff2190c0000000000000000000000000a636b706f6f6c1365752f736f6c6f2e636b706f6f6c2e6f72672fffffffff037a7c93120000000016001480bded37e3f86a1a546e099b15e2b02855d3843cfd0c61000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed2aaf372eab3aa252e8b72245d385e68a932bf06cf5c06ed51d013f6599d62d5b00000000
```

Decode that as you would normally decode a Bitcoin tx ([online decoder](https://www.blockchain.com/explorer/assets/btc/decode-transaction)) to get the script sig (tx input script). In this case the script sig is:
```
0392ca0d0004b27c666804e66ff2190c0000000000000000000000000a636b706f6f6c1365752f736f6c6f2e636b706f6f6c2e6f72672f
```
The height is at the front of this script:

The first byte (0x03) is a push opcode: push next 3 bytes

Next 3 bytes (0x92ca0d) is the height in little-endian

```
0x92 = 146
0xca = 202
0x0d = 13
```
Converting from little-endian: 146 + (202 × 256) + (13 × 256²) = 903826 <- the height

There's other stuff in this script sig too, like the ASCII script tag:
```
eu/solo.ckpool.org/
```
which would probably also be good to display in the firmware.

I think that showing all of this data would be great because it brings transparency and as skot mentioned, it would be an awesome real-time learning tool that no other mining hardware is doing.

skot, 2025-08-20:
I think it should be on the main AxeOS tab.. maybe a new block header info card?

mutatrum, 2025-08-20:
Can start there. I would like to extend it to its own tab, with history of jobs and best diffs per job, and a proper header breakdown. IMO that's a good education tool.

mutatrum, 2025-08-20:
A new card is quite involved, as it will be a whole new line of cards.

[image]

PR #1200 (2025-09-13): Show block header, scriptsig and network difficulty
Replaced the Uptime card with Block Header:

[image]

Same info on the screen:

[image]

If people still want the Uptime card, that can be restored. But IMO another row of cards is a bit much, or we have to condense them further.

Fixes #1011
Fixes #1166
commit 3a063f6b (2025-09-13): Show block header, scriptsig and network difficulty (#1200)

* Add block header and miner tag

Fixes #1011
Fixes #1166

* Add network difficulty to Mining card

* EOF whitespace

* Add mock values

* Change card title

* Read complete scriptsig

* Organised code

* Fix baseline

* Add Block Header info to screen

* Renamed miner_tag to scriptsig

* Code review fixes

* Remove commented out code

* Fix borked-up merge
closed 2025-09-13 by WantClue