bitaxeorg/ESP-Miner #1031: Bitaxe Gamma not mining

vekexasia, 2025-06-12:
I am using a bitaxe gamma and cant understand the reason why it does not mine.

This is the log

```
I (26) boot: ESP-IDF v5.4.1-dirty 2nd stage bootloader
I (26) boot: compile time May  5 2025 23:04:07
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00400000
I (78) boot:  3 www              Unknown data     01 82 00410000 00300000
I (84) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (91) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (97) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (104) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr=00710020 vaddr=3c0e0020 size=30278h (197240) map
I (156) esp_image: segment 1: paddr=007402a0 vaddr=3fc9be00 size=05390h ( 21392) load
I (161) esp_image: segment 2: paddr=00745638 vaddr=40374000 size=0a9e0h ( 43488) load
I (171) esp_image: segment 3: paddr=00750020 vaddr=42000020 size=d1cd4h (859348) map
I (323) esp_image: segment 4: paddr=00821cfc vaddr=4037e9e0 size=0d3f4h ( 54260) load
I (335) esp_image: segment 5: paddr=0082f0f8 vaddr=600fe100 size=0001ch (    28) load
I (344) boot: Loaded app from partition at offset 0x710000
I (345) boot: Disabling RNG early entropy source...
I (355) octal_psram: vendor id    : 0x0d (AP)
I (355) octal_psram: dev id       : 0x02 (generation 3)
I (356) octal_psram: density      : 0x03 (64 Mbit)
I (360) octal_psram: good-die     : 0x01 (Pass)
I (365) octal_psram: Latency      : 0x01 (Fixed)
I (371) octal_psram: VCC          : 0x01 (3V)
I (376) octal_psram: SRF          : 0x01 (Fast Refresh)
I (382) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (387) octal_psram: BurstLen     : 0x01 (32 Byte)
I (393) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (399) octal_psram: DriveStrength: 0x00 (1/1)
I (405) MSPI Timing: PSRAM timing tuning index: 5
I (410) esp_psram: Found 8MB PSRAM device
I (414) esp_psram: Speed: 80MHz
I (418) cpu_start: Multicore app
I (835) esp_psram: SPI SRAM memory test OK
I (844) cpu_start: Pro cpu start user code
I (844) cpu_start: cpu freq: 240000000 Hz
I (844) app_init: Application information:
I (847) app_init: Project name:     esp-miner
I (852) app_init: App version:      4457296
I (857) app_init: Compile time:     Jun  4 2025 10:51:01
I (863) app_init: ELF file SHA256:  3bf993d19...
I (868) app_init: ESP-IDF:          v5.4.1
I (873) efuse_init: Min chip rev:     v0.0
I (878) efuse_init: Max chip rev:     v0.99
I (883) efuse_init: Chip rev:         v0.2
I (888) heap_init: Initializing. RAM available for dynamic allocation:
I (895) heap_init: At 3FCB71E0 len 00032530 (201 KiB): RAM
I (901) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (907) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (913) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (919) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (928) spi_flash: detected chip: generic
I (931) spi_flash: flash io: dio
I (936) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (943) sleep_gpio: Enable automatic switching of GPIO sleep configuration
[... 269 lines cut ...]
I (16041) bm1370Module: Setting Frequency to 593.75MHz (593.75)
I (16141) bm1370Module: Setting Frequency to 600.00MHz (600.00)
I (16241) bm1370Module: Setting Frequency to 606.25MHz (606.25)
I (16341) bm1370Module: Setting Frequency to 612.50MHz (612.50)
I (16441) bm1370Module: Setting Frequency to 618.75MHz (618.75)
I (16541) bm1370Module: Setting Frequency to 625.00MHz (625.00)
I (16641) bm1370Module: Setting Frequency to 631.25MHz (631.25)
I (16741) bm1370Module: Setting Frequency to 636.00MHz (635.71)
I (16841) bm1370Module: Setting Frequency to 636.00MHz (635.71)
I (16841) frequency_transition: Successfully transitioned ASIC type 1370 to 636.00 MHz
I (16841) bm1370Module: Setting max baud of 1000000
I (16851) serial: Changing UART baud to 1000000
I (16851) stratum_task: Opening connection to pool: eu.stratum.braiins.com:3333
I (16851) stratum_task: Starting heartbeat thread for primary pool: eu.stratum.braiins.com:3333
I (16861) ASIC_task: ASIC Job Interval: 500.00 ms
I (16871) stratum_task: Connecting to: stratum+tcp://eu.stratum.braiins.com:3333 (164.92.209.20)
I (16871) ASIC_task: ASIC Ready!
I (16891) stratum_task: Socket created, connecting to 164.92.209.20:3333
I (16861) statistics_task: Starting
I (16901) statistics_task: Disabled!
I (16861) main_task: Returned from app_main()
I (16951) stratum_task: Resetting stratum uid
I (16951) stratum_task: Clean Jobs: clearing queue
I (16951) stratum_api: tx: {"id": 1, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}?
```

vekexasia, 2025-06-12:
I also tried with a bench psu to set exactly 5.0v (because i saw the original PSU had some high voltage).

nothing has changed. bench psu reports ~0.2A of consumption after startup.

skot, 2025-06-12:
Hi there, I'm sorry you are having trouble with your Bitaxe. This is not a tech support forum. You can contact the seller where you purchased the Bitaxe for support, or if you built it yourself ask on the OSMU Discord. https://osmu.xyz