bitaxeorg/ESP-Miner #1052: Feature Request: Display HOSTNAME in <title> and top of AxeOS page

homecryptominer, 2025-06-20:
Not a bug, but to help improve UX. There are many people out there with many Bitaxes. When having multiple tabs opened in a browser, hovering over the tab just shows AxOS and the IP. It would be really useful to also include the actual HOSTNAME in the <title> tag.

In addition to this, it would also be useful to render the HOSTNAME at the top of the AxeOS Dashboard somewhere - perhaps to the right hand side of the hamburger menu?

Thank you so much for your continued support team! :-)
Steve

duckAxe, 2025-06-20:
Maybe like that. Hostname & ASIC added to the page header. Mobile hostname only.

[image]

[image]

duckAxe, 2025-06-20:
I would prefer the following solution: We should show the hostname and Wi-Fi signal strength because that information is much more important and helpful than the ASIC.

On desktop, we display this information in the top right corner. Since there isn't enough space for it on mobile, we display it in the menu.

**Desktop**

https://github.com/user-attachments/assets/9935fa36-8d68-4272-aad3-bd88a3981d67

**Mobile**

https://github.com/user-attachments/assets/59b71a39-38e7-4482-9bff-c03a315b7e27

mutatrum, 2025-06-20:
I like that.

skot, 2025-06-20:
I like this too. My only suggestion is that the icon for hostname should be something looks like a bitaxe and not a computer (since it's the Bitaxe hostname)

duckAxe, 2025-06-20:
@skot We use [PrimeNg Icons](https://primeng.org/icons#list). There are not many icons that match this.

skot, 2025-06-20:
hmm, yeah no ideal icons in there. If we really can't have a Bitaxe icon, then maybe one of these?

[image]

[image]

[image]

[image]

[image]

duckAxe, 2025-06-20:
Check this. 

[image]

Found on bitaxe.org 

[image]

mutatrum, 2025-06-20:
Made a quick and dirty outline of the above image. Maybe this can be used as a custom icon?

[image]

duckAxe, 2025-06-20:
@mutatrum Your svg is good, but not optimal for a very small icon. I made a clean pixel art icon that looks good as a tiny icon. What do you think?

[image]

**Yours**
[image]

**Mine**
[image]

mutatrum, 2025-06-20:
Can you see if you get this working?
```
    <ng-template #bitaxeicon>
        <svg xmlns="http://www.w3.org/2000/svg" viewBox="0 0 24 24">
            <g id="bitaxe">
                <path d="M5 3A1 1 0 004 4V25a1 1 0 001 1H20a1 1 0 001-1V4A1 1 0 0020 3H17V2A1 1 0 0016 1H9A1 1 0 008 2V3ZM5 4H8V5a1 1 0 001 1h7a1 1 0 001-1V4h3V8H17v2h3v1H19v1h1V25H5Zm4-2h7V5H9ZM7 18a1 1 0 0111 0A1 1 0 017 18Zm1 0a1 1 0 009 0 1 1 0 00-9 0ZM9 8H7A1 1 0 006 9v2a1 1 0 001 1H9a1 1 0 001-1V9A1 1 0 009 8ZM9 9v2H7V9Z"/>
            </g>
        </svg>
    </ng-template>
```

duckAxe, 2025-06-21:
Inline embedding will work, but we can't use it because we embed the SVG file in several places. That is why I have saved it as an image file, see my PR.

Let me know if you want me to change anything else.

mutatrum, 2025-06-21:
That's not bad for a few minutes fiddling 😆 
[image]

FYI, I used this to make it: https://yqnn.github.io/svg-path-editor/

duckAxe, 2025-06-21:
Awesome tool. Props to you.

homecryptominer, 2025-06-22:
You guys are BRILLIANT! I love the changes. This change with the HOSTNAME also in the web <title> bar will help me quickly identify which Bitaxe I am looking at without having to scroll to the bottom of the screen to check my pool where I have my BTC-ADDRESS.HOSTNAME.