bitaxeorg/ESP-Miner #1057: Icons not rendering properly on v2.9.0b2

Hylinus, 2025-06-21:
Icons such as the ones that appear to the left of the Menu options are not rendering properly on Chromium- or Firefox-based browsers.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Dashboard'
2. Notice that the icons that should appear to the left of the text description for the options under Menu are not rendering (Chromium-based browsers) or appear as a small square with "E9" on top and a what appears to be a 2-digit hex number on the bottom (Firefox-based browsers)

**Expected behavior**
Icons such as the "Swarm", "Network", "Customization" or the sort icon in the Swarm page should render like in prior versions.

**Screenshots & Photos**
v2.9.0b2
[image]

v2.6.3
[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma 601
 - Bitaxe HW vendor: Altair Tech
 - ESP-Miner FW version: v2.9.0b2

0xf0xx0, 2025-06-21:
fixed in the next beta

duckAxe, 2025-06-22:
@Hylinus https://github.com/bitaxeorg/ESP-Miner/releases/tag/v2.9.0b3

Hylinus, 2025-06-22:
Icons rendering as expected on all affected pages. Thank you both!