bitaxeorg/ESP-Miner #1061: On Web Reset only: asci status chip count 0

harvybob, 2025-06-23:
Note: Issues are not for customer support, configuration or discussion. For those topics please consult with your HW vendor or the OSMU Discord at: https://osmu.bitaxe.org

**Describe the bug**
When selecting web restart of miner, Display shows "asci status chip count 0". If pressing reset button or power down/back up, miner functions as normal.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to standard homepage
2. Click on restart
4. See error "asci status chip count 0" on display

**Expected behavior**
Miner restarts and continues mining (1 of mine does, the other does not)

**Screenshots & Photos**
If applicable, add AxeOS screenshots and/or photos of your Bitaxe to help explain your problem.

**Hardware (please complete the following information):**
 - Bitaxe HW version: Gamma 601
 - Bitaxe HW vendor: bitaxeshop
 - ESP-Miner FW version: [e.g. 2.1.1, etc] - v2.8.1 (also happened on v2.8.0) ,ESP-IDF Version:  v5.4.1 
 - Hash Frequency: 525
 - Voltage: 1150
 - Pool URL, Port, User: eu.stratum.braiins.com

**Additional context**
Believe this is a hardware problem, as same firmware is running on two Gamma 601's with different behaviour. my "Alpha" restarts fine and back to hashing, the "Beta" comes up with the error.
Both running the same hardware, both have the same stock settings, both mining to the same pool.

**Looking for any info/guidance on what to look for to try and identify and resolve the problem.**
Bitaxeshop are already sending out a replacement, but more wanting to understand why it doesnt restart.

I have lapped and re thermal pasted the stock heatsinks - temp drop from approx 70-72 to 65 degrees (its hot in the UK at the moment!), waiting for 3d print mounts for ice towers. Issue happened before and after these changes.

harvybob, 2025-06-23:
> Could you get a USB log from the failing bitaxe including when you press the AxeOS restart button up to when you see the error on the screen?
> 
> Are these GekkoScience Edition Bitaxes?

is there a set of instructions on how to get the usb log? (i presume if i use the IP I'm just getting the web one?)
Not sure re GekkoScience Edition Bitaxes - I cant see any specific markings on it other then Bitaxe in gold, and 601 in silkscreen on a black board - are there any obvious hardware differences that I can compare images from to my actual hardware?

skot, 2025-06-24:
@harvybob what brand power supplies are you using? Same power supply make and model on both Alpha and Beta?

A picture of the PSU would be helpful.